lvgl/lv_port_pc_eclipse #1: Can't import to eclipse

KcMeterCEC, 2017-05-14:
Hi, your GUI is very fascinating , so i want to explore it.

Then i follow your guide[ step by step](http://littlev.hu/index.php/getting-started/run-on-pc/) .

When i want to import project to eclipse, **the eclipse warn can't find project file!**

And i find your ".gitignore" file is ignored ".cproject" and ".settings" , i think it's why i can't import...

Please fix the bug.Thanks!

kisvegabor, 2017-05-15:
Hi!

I removed those files from .gitignore and pushed them. Thank you for reporting the issue.

Today I started a new website with and released an improved version of the library. I recommend you to use the new versions: http://gl.littlev.hu

KcMeterCEC, 2017-05-15:
Thanks!

Your new website is awesome!

kisvegabor, 2017-05-15:
Thanks. I hope it is useful too :)

Please keep me informed if you have any other problem with the PC simulator.

closed 2017-11-20 by kisvegabor