lvgl/lv_port_pc_eclipse #2: Suggestion: Visual studio support

ramonschepers, 2017-05-28:
Is it possible to support visual studio without sdl library? I know it's possible, just need so effort to port it.

A example project including simulator project is here https://github.com/rossumur/microtouch

This is originally for an atmega32u4, but since it's done very simple, it's easy to adapt! :)

Why am I suggesting this?
To get people to dev easier on the PC simulator by just a few steps to undertake.

kisvegabor, 2017-05-29:
Thanks you for this information. I will check and try it soon!

ramonschepers, 2017-05-29:
You can even build a executable from withing visual studio so people can try it themselves without compiling ;)

kisvegabor, 2017-05-29:
Good idea. However it won't be cross-platform but it might be useful for a lot of people.

TheGrovesy, 2018-05-19:
Has this been taken any further? Getting the simulation buildable from visual studio would be really useful to us.

kisvegabor, 2018-05-20:
Hi,

However there is no an out of the box Visual studio project so far some people already confirmed that the library can be built in Visual Studio.

I just releases v5.1.1 where you can disable the use of Variable Length Arrays in lv_conf.h which was the only not supported thing in Visual Studio. (`LV_COMPILER_VLA_SUPPORTED`).

kisvegabor, 2018-05-22:
Awesome! Thank you very much!

TheGrovesy, 2018-06-11:
Sorry i completely forgot to share this. For now its a complete zip file containing the Visual Studio solution, the LittlevGL source, the SDL binaries and a few extra files which are needed to get around the need for mingW. At some point I plan to create a github repo that links to the LittlevGL source but not sure how yet (new to github).
[Here is the link](https://1drv.ms/u/s!Aj3GgbbKuEhYgqs1OpLrJT68r0T9DQ)
There is a ReadMe file within the Visual Studio folder where I have defined what I did to get this working. I also had to change some of the littlevGL files and have a few suggestions as follows:

Changes:
> main.c
	> Added #include <windows.h>
	> Changed usleep(1000) to Sleep(1)

> main.c, mouse.c, monitor.c, keyboard.h
	> Change #include "SDL2/SDL.h" to #include "SDL2/include/SDL.h"

> pc simulator/lv_conf.h
    > change line 85 '#define LV_COMPILER_VLA_SUPPORTED    1' to a zero '0'

> Had to remove lv_tests folder from the project due to 'C2099 Error Initializer is not a constant' Regarding the use of macro LV_COLOR_XXX
>

Notes/Suggestions to Author:
> Maybe add a '#ifndef WIN32' around '#define LV_COMPILER_VLA_SUPPORTED 1' and define WIN32 with a zero.
> strcpy is considered unsafe so I had to disable deprecation using _CRT_SECURE_NO_WARNINGS in Properties > C/C++ > Preprocessor > Preprocessor Definitions
> Suggest changing the header includes for SDL2/SDL.h to SDL2/include/SDL.h

Let me know if you have any issues or comments. Thanks.

kisvegabor, 2018-06-12:
Hi,

Thank you  very much!
I'm on Linux and unfortunately I didn't used Visual Studio so far.  However I just installed Windows in Virtual Box so will try the project soon!

TheGrovesy, 2018-06-13:
Great, I would be grateful to have any feedback from you. 

My next plan will be to set this up through Github and link to your repo so any updates you make can be easily tested... I just need to work out Git... its quite different from SVN which I am used to. Any tips on linking to your repo would be create.

kisvegabor, 2018-06-13:
I suggest this reading: https://guides.github.com/activities/forking/

kisvegabor, 2018-07-13:
Hi,

I tested your project in Visual Studio 2017.

I needed to "Retarget project" (Right click on project > Retarget project) to find the standard header files.

After it, the compiler didn't found the SDL related functions. For example:
`LNK2019	unresolved external symbol SDL_CreateThread referenced in function monitor_init`

Have you also used Visual Studio 2017? Or do you have any idea?

kisvegabor, 2018-07-14:
@TheGrovesy Meanwhile I figured out how to create a Visual Studio project with SDL from scratch. This video tutorial helped a lot: https://www.youtube.com/watch?v=Sfn7yOiwJLw

I followed your idea to copy SDL into the project so no extra copies of library files are required.
I also added some new config options to the `lvgl`, `lv_examples`, `lv_drivers` to support Visual Studio out of the box.

Here is the project: https://github.com/littlevgl/lv_projects/tree/master/visual_studio_2017_sdl_x64 

So after near one year a close this issue :)

If you find bugs or have remarks, please open an ew issue for them!

Thank you vers much for your contribution! :)

closed 2018-07-14 by kisvegabor