lvgl/lv_port_pc_eclipse #3: ../hw/per/psp/pc/psp_tmr.c:13:21: fatal error: pthread.h: No such file or directory

glowlabs, 2017-07-28:
Hi, this looks like a great library. I'm having an issue when I try and build:

17:43:43 ** Incremental Build of configuration Debug for project lv_pc **
make all 
'Building file: ../hw/per/psp/pc/psp_tmr.c'
'Invoking: Cross GCC Compiler'
gcc -I"C:\Users\Josh\Documents\proj_pc" -O0 -g3 -Wall -c -fmessage-length=0 -MMD -MP -MF"hw/per/psp/pc/psp_tmr.d" -MT"hw/per/psp/pc/psp_tmr.o" -o "hw/per/psp/pc/psp_tmr.o" "../hw/per/psp/pc/psp_tmr.c"
../hw/per/psp/pc/psp_tmr.c:13:21: fatal error: pthread.h: No such file or directory
 #include <pthread.h>
                     ^
compilation terminated.
hw/per/psp/pc/subdir.mk:21: recipe for target 'hw/per/psp/pc/psp_tmr.o' failed
make: * [hw/per/psp/pc/psp_tmr.o] Error 1

17:43:44 Build Finished (took 947ms)

kisvegabor, 2017-07-28:
It seems you are using  windows. In this case you nee to download pthread.dll too.
http://sourceware.org/pthreads-win32/

kisvegabor, 2017-07-28:
I tried it on Windows again and need to do the following steps:
1. Download **pthread** from here: ftp://sourceware.org/pub/pthreads-win32/pthreads-w32-2-9-1-release.zip
2. Copy *Pre-built.2/include/* content to *C:/MinGW/include*
3. Copy *Pre-built.2/lib/x64/* content to *C:/MinGW/lib* (If you have a 64 machine)
4. Copy *Pre-built.2/dll/x64/* content to *{eclipse_worksapce}/lv_pc/Debug/*
5. I got an error: 'struct timespec' redefined. To solve this: *Project properties -> C/C++ Build -> Settings -> GCC C++ Compiler -> Miscellaneous -> Others flag* and add  **-DHAVE_STRUCT_TIMESPEC** (it could be a pthread issue)
6. Download the development libraries of SDL from https://www.libsdl.org/download-2.0.php here *Development Libraries: SDL2-devel-2.0.5-mingw.tar.gz* 
To get it work I need to use the *i686* (32 bit) libraries even on 64 bit system. So if you get *undefined reference to SDL_init()* error try the *x86_64_w64_mingw* folder
7. Copy *i686-w64-mingw32/include/SDL2* folder to *C:/MinGW/include*
8. Copy *i686-w64-mingw32/lib/* content to *C:/MinGW/lib* 
9. Copy *i686-w64-mingw32/bin/SDL2.dll* content to *{eclipse_worksapce}/lv_pc/Debug/*

Does it working now?

kisvegabor, 2017-12-21:
In v5.0.0 pthread is already not used. [More info](http://www.gl.littlev.hu/pc-simulator)

closed 2017-12-21 by kisvegabor