lvgl/lv_port_pc_eclipse #4: About the task's schedule period ?

KcMeterCEC, 2017-09-19:
Hi, i have a question about the task's schedule period.

I have seen your code about function :  **ptask_exec**. 

Then i understood that the tasks will be execute every **ptask_p->period** times in millisecond.

But i also have seen your **main** function code:
``` c   
while(1) {
        /* Periodically call the ptask handler.
         * It could be done in a timer interrupt or an OS task too.*/
        ptask_handler();
        usleep(5000);

    }
```
When i use the function: **ptask_create** to create a task, and parameter two is 5ms.

**It seems like that the task will be execute every 25ms**.

Did i wrong?

Why you use delay function at there?

kisvegabor, 2017-09-19:
Hi,

It should be called in every 5 ms if the period is 5 ms regardless to the delay after `ptask_handler()` because it uses the *systick* from *hal* to  get the elapsed time.

The delay is applied to not call *systick* functions at a very high rate because depending on your implementation it can disable interrupts when getting the current tick count.

Are you sure the call period is 25 ms in your case?
Dose something happen is you write `usleep(1000);` ?

KcMeterCEC, 2017-09-19:
Oh!
It is my fault.I didn't really test the call period.
So, if i use **usleep(5000)** after **ptask_handler()**, inorder to keep in time, the task period couldn't set less than 5ms.
Thanks a lot !

kisvegabor, 2017-09-19:
Yes, as you say!

Your very welcome!

closed 2017-09-19 by kisvegabor