lvgl/lv_port_pc_eclipse #5: problems in building the project by eclipse in Mac OS X

jason-sophia, 2018-01-11:
build the project by eclipse in Mac OS X ,build error:
ld: library not found for -lSDL2
clang: error: linker command failed with exit code 1 (use -v to see invocation)

I have already install SDL ,Copy the SDL2.framework to /Library/Frameworks
Could you help me

kisvegabor, 2018-01-11:
Unfortunately I'm not an OSX user and I don't know how to link libraries to your project. You should consult someone who familiar with development under OSX. However once you link SDL to your project it should work because SDL is cross-platform.

If you succeed please inform me because I would share the steps on my webpage.

margce, 2018-04-03:
Hi, I just started testing Littlevgl. I had the same issue as Jason, in order to succesfully compile on MacOS you need to download the SDL2 library and then configure Eclipse to as follows.

OPTION 1. Using SDL2 libraries directly from SDL2 website.
1.  Need to include SDL headers
Open the pc_simulator project properties and go to C/C++ Build -> Settings, under Cross GCC Compiler select Includes and add the following /Library/Frameworks/SDL2.framework/Versions/A/Headers

2. You need to set configure the linker
Under the same screen select Cross GCC Linker -> Miscellaneous and add the following
-F/Library/Frameworks -framework SDL2 -framework Cocoa

Then select Libraries and remove the SDL2main and SDL2 

OPTION2. Using SDL2 libraries from Brew.
1. Using brew install SDL2 from a terminal using
brew install sdl2
2. After installation is complete you should be able to compile successfully from Eclipse, but Brew libraries may not be up to date compared with SDL2 original source.

Now that you can compile, you will face another issue under MacOS I will report that on a different issue.

kisvegabor, 2018-04-03:
Thank you @margce . 
As LittlevGL uses only the very basic features of SDL it's not problem if we don't use the latest version. I added these steps on the website to help others.

As OSX compatibility seems to be solved I close this issue.
Thank you very much  for the contribution!

closed 2018-04-03 by kisvegabor