lvgl/lv_port_pc_eclipse #6: NEED FOR HELP

DengQilong, 2018-01-14:
Hello,boys:
I used eclipse CDT to run this code,but got an error.
"../main.c:44:5: error: conflicting types for 'SDL_main'
 int main (void)"
How to solve this problem?

kisvegabor, 2018-01-14:
Hi,
Are you using Windows, Linux or OSX?

I found this as a possible solution: add `_SDL_main_h` define in Project properties -> C/C++ build ->Settings -> Preprocessor ->Defined symbol (-D)

DengQilong, 2018-01-15:
I use Windows.

kisvegabor, 2018-01-15:
And have you tried the mentioned fix?

DengQilong, 2018-01-15:
I will try it in the evening.And I will give the feedback.

kisvegabor, 2018-01-21:
I see. Thank you for the feedback.
I found and other possible solution (because it works for me i can't try it): add argc and arv to the main function.
`int main(int argc, char** argv)`

DengQilong, 2018-01-28:
Dear brother,
I got more errors.This is the log...
19:54:20 **** Build of configuration Debug for project pc_simulator ****
make all 
'Building target: pc_simulator'
'Invoking: Cross GCC Linker'
gcc  -o "pc_simulator"  ./lvgl/lv_themes/lv_theme.o ./lvgl/lv_themes/lv_theme_alien.o ./lvgl/lv_themes/lv_theme_default.o ./lvgl/lv_themes/lv_theme_material.o ./lvgl/lv_themes/lv_theme_mono.o ./lvgl/lv_themes/lv_theme_night.o ./lvgl/lv_themes/lv_theme_templ.o ./lvgl/lv_themes/lv_theme_zen.o  ./lvgl/lv_objx/lv_bar.o ./lvgl/lv_objx/lv_btn.o ./lvgl/lv_objx/lv_btnm.o ./lvgl/lv_objx/lv_cb.o ./lvgl/lv_objx/lv_chart.o ./lvgl/lv_objx/lv_cont.o ./lvgl/lv_objx/lv_ddlist.o ./lvgl/lv_objx/lv_gauge.o ./lvgl/lv_objx/lv_img.o ./lvgl/lv_objx/lv_kb.o ./lvgl/lv_objx/lv_label.o ./lvgl/lv_objx/lv_led.o ./lvgl/lv_objx/lv_line.o ./lvgl/lv_objx/lv_list.o ./lvgl/lv_objx/lv_lmeter.o ./lvgl/lv_objx/lv_mbox.o ./lvgl/lv_objx/lv_objx_templ.o ./lvgl/lv_objx/lv_page.o ./lvgl/lv_objx/lv_roller.o ./lvgl/lv_objx/lv_slider.o ./lvgl/lv_objx/lv_sw.o ./lvgl/lv_objx/lv_ta.o ./lvgl/lv_objx/lv_tabview.o ./lvgl/lv_objx/lv_win.o  ./lvgl/lv_misc/lv_fonts/dejavu_10.o ./lvgl/lv_misc/lv_fonts/dejavu_10_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_10_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_10_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_10_sup.o ./lvgl/lv_misc/lv_fonts/dejavu_20.o ./lvgl/lv_misc/lv_fonts/dejavu_20_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_20_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_20_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_20_sup.o ./lvgl/lv_misc/lv_fonts/dejavu_30.o ./lvgl/lv_misc/lv_fonts/dejavu_30_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_30_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_30_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_30_sup.o ./lvgl/lv_misc/lv_fonts/dejavu_40.o ./lvgl/lv_misc/lv_fonts/dejavu_40_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_40_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_40_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_40_sup.o ./lvgl/lv_misc/lv_fonts/dejavu_60.o ./lvgl/lv_misc/lv_fonts/dejavu_60_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_60_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_60_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_60_sup.o ./lvgl/lv_misc/lv_fonts/dejavu_80.o ./lvgl/lv_misc/lv_fonts/dejavu_80_cyrillic.o ./lvgl/lv_misc/lv_fonts/dejavu_80_latin_ext_a.o ./lvgl/lv_misc/lv_fonts/dejavu_80_latin_ext_b.o ./lvgl/lv_misc/lv_fonts/dejavu_80_sup.o ./lvgl/lv_misc/lv_fonts/symbol_10_basic.o ./lvgl/lv_misc/lv_fonts/symbol_10_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_10_file.o ./lvgl/lv_misc/lv_fonts/symbol_20_basic.o ./lvgl/lv_misc/lv_fonts/symbol_20_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_20_file.o ./lvgl/lv_misc/lv_fonts/symbol_30_basic.o ./lvgl/lv_misc/lv_fonts/symbol_30_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_30_file.o ./lvgl/lv_misc/lv_fonts/symbol_40_basic.o ./lvgl/lv_misc/lv_fonts/symbol_40_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_40_file.o ./lvgl/lv_misc/lv_fonts/symbol_60_basic.o ./lvgl/lv_misc/lv_fonts/symbol_60_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_60_file.o ./lvgl/lv_misc/lv_fonts/symbol_80_basic.o ./lvgl/lv_misc/lv_fonts/symbol_80_feedback.o ./lvgl/lv_misc/lv_fonts/symbol_80_file.o  ./lvgl/lv_misc/lv_anim.o ./lvgl/lv_misc/lv_area.o ./lvgl/lv_misc/lv_circ.o ./lvgl/lv_misc/lv_color.o ./lvgl/lv_misc/lv_font.o ./lvgl/lv_misc/lv_fs.o ./lvgl/lv_misc/lv_ll.o ./lvgl/lv_misc/lv_mem.o ./lvgl/lv_misc/lv_task.o ./lvgl/lv_misc/lv_templ.o ./lvgl/lv_misc/lv_trigo.o ./lvgl/lv_misc/lv_txt.o ./lvgl/lv_misc/lv_ufs.o  ./lvgl/lv_hal/lv_hal_disp.o ./lvgl/lv_hal/lv_hal_indev.o ./lvgl/lv_hal/lv_hal_tick.o  ./lvgl/lv_draw/lv_draw.o ./lvgl/lv_draw/lv_draw_rbasic.o ./lvgl/lv_draw/lv_draw_vbasic.o  ./lvgl/lv_core/lv_group.o ./lvgl/lv_core/lv_indev.o ./lvgl/lv_core/lv_obj.o ./lvgl/lv_core/lv_refr.o ./lvgl/lv_core/lv_style.o ./lvgl/lv_core/lv_vdb.o  ./lv_examples/lv_tutorial/9_responsive/lv_tutorial_responsive.o  ./lv_examples/lv_tutorial/8_animations/lv_tutorial_animations.o  ./lv_examples/lv_tutorial/7_fonts/lv_tutorial_fonts.o ./lv_examples/lv_tutorial/7_fonts/ubuntu_40_ascii.o ./lv_examples/lv_tutorial/7_fonts/ubuntu_40_cyrillic.o  ./lv_examples/lv_tutorial/6_images/img_red_flower.o ./lv_examples/lv_tutorial/6_images/lv_tutorial_images.o  ./lv_examples/lv_tutorial/5_antialiasing/lv_tutorial_antialiasing.o  ./lv_examples/lv_tutorial/4_themes/lv_tutorial_themes.o  ./lv_examples/lv_tutorial/3_styles/lv_tutorial_styles.o  ./lv_examples/lv_tutorial/2_objects/lv_tutorial_objects.o  ./lv_examples/lv_tutorial/1_hello_world/lv_tutorial_hello_world.o  ./lv_examples/lv_tutorial/10_keyboard/lv_tutorial_keyboard.o  ./lv_examples/lv_tutorial/0_porting/lv_tutorial_porting.o  ./lv_examples/lv_tests/lv_test_theme/lv_test_theme.o  ./lv_examples/lv_tests/lv_test_stress/lv_test_stress.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_win/lv_test_win.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_tabview/lv_test_tabview.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_ta/lv_test_ta.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_sw/lv_test_sw.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_slider/lv_test_slider.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_roller/lv_test_roller.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_page/lv_test_page.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_mbox/lv_test_mbox.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_lmeter/lv_test_lmeter.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_list/lv_test_list.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_line/lv_test_line.o  ./lv_examples/lv_te [... cut ...]
G:\MyC++WorkSpace\pc_simulator\Debug/../lv_drivers/display/monitor.c:251: undefined reference to `SDL_Quit'
./main.o: In function `hal_init':
G:\MyC++WorkSpace\pc_simulator\Debug/../main.c:97: undefined reference to `SDL_CreateThread'
./main.o: In function `tick_thread':
G:\MyC++WorkSpace\pc_simulator\Debug/../main.c:104: undefined reference to `SDL_Delay'
collect2.exe: error: ld returned 1 exit status
makefile:81: recipe for target 'pc_simulator' failed
make: *** [pc_simulator] Error 1

19:54:24 Build Finished (took 3s.598ms)

kisvegabor, 2018-01-28:
Hi,
it's fine because now the original issue is solved.

Now the issue is SDL is not linked properly. Are you sure you use 64 bit MinGW and 64 bit SDL? (or 32 bit both)

DengQilong, 2018-01-28:
I use 32 bit both

DengQilong, 2018-01-28:
There are only two error.
lvgl/lv_misc/lv_fonts/subdir.mk:159: recipe for target 'lvgl/lv_misc/lv_fonts/symbol_40_feedback.o' failed
make: *** [lvgl/lv_misc/lv_fonts/symbol_40_feedback.o] Error 1

kisvegabor, 2018-01-29:
And why does it fail? Can you attache a longer compiler log?

And what solved the previous issue?

kisvegabor, 2018-08-10:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-08-10 by kisvegabor

shb3014, 2024-01-18:
run into this today and solved by adding target_compile_options(main PRIVATE -Dmain=SDL_main) to cmake