lvgl/lv_port_pc_eclipse #7: How to make  the pc to simulate the monochrome LCD

qwert1213131, 2018-02-02:
Hi,
Need to modify some  source files?

kisvegabor, 2018-02-02:
Just set `LV_COLOR_DEPTH  1` in lv_conf.h.

When I tried I noticed that img_benchmark_bg.c is not updated handle i bit color depth.
To fix it modify  `#if LV_COLOR_DEPTH == 8` to `#if LV_COLOR_DEPTH == 8 || LV_COLOR_DEPTH == 1`

qwert1213131, 2018-02-05:
I set the LV_COLOR_DEPTH 1，there is still no change in color。
Does this have to do with the theme, do I need to set the theme for mono?
[image]
[image]

kisvegabor, 2018-02-05:
Strange. It works for me. Are you sure the project is re-compiled?

qwert1213131, 2018-02-06:
hi,
I'll test it on another Linux computer. Thank you！

qwert1213131, 2018-02-06:
[image]
Thanks!
It's worked!
May be the wrong QT IDE setting to casuse it.

kisvegabor, 2018-02-06:
Great! :)

closed 2018-02-06 by kisvegabor