lvgl/lv_port_pc_eclipse #10: Build instructions with Eclipse produce two errors

robtow, 2018-03-07:
**Following these instructions from README.md:**

   Import the PC simulator project
    Open Eclipse CDT
    Click File->Import and choose General->Existing project into Workspace
    Browse the root directory of the project and click Finish
    Build your project and run it

**Results in two errors:**

Errors (2 items)
Description	Resource	Path	Location	Type
fatal error: SDL2/SDL.h: No such file or directory	keyboard.h	/lv_dev_5/lv_drivers/indev	line 20	C/C++ Problem
make: *** [lv_examples/lv_tests/lv_test_group/lv_test_group.o] Error 1	lv_dev_5		 	C/C++ Problem

kisvegabor, 2018-03-07:
A few questions:
- Which operation system  do you use?
- Just to be sure: have you installed SDL2?
- Are sure you have chosen the correct architecture for SDL2? (must be same as your compiler's: 32 bit or 64 bit)

kisvegabor, 2018-03-08:
We have discussion about compiling with SDL on Windows. IF you are also on Windows pleas check it: https://github.com/littlevgl/lvgl/issues/123#issuecomment-371322117

kisvegabor, 2018-08-10:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-08-10 by kisvegabor