lvgl/lv_port_pc_eclipse #12: Workaround to run under MacOS

margce, 2018-04-03:
Hi,

I just started using the PC_SIMULATOR under MacOS 10.12.6 and I ran into the following issue.

Using the default source code when running the pc_simulator application the app crashes with the following message

_Martins-MacBook-Pro:Debug tinno$ ./pc_simulator
2018-04-03 10:17:46.692 pc_simulator[4326:483915] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fffcdaaf2cb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffe28c648d objc_exception_throw + 48
	2   AppKit                              0x00007fffcbc98e82 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 4480
	3   SDL2                                0x00000001056195f5 SDL2 + 140789
	4   SDL2                                0x00000001056038dd SDL2 + 51421
	5   pc_simulator                        0x00000001053faa5b sdl_refr + 395
	6   SDL2                                0x00000001056176ac SDL2 + 132780
	7   SDL2                                0x0000000105617139 SDL2 + 131385
	8   libsystem_pthread.dylib             0x00007fffe33c593b _pthread_body + 180
	9   libsystem_pthread.dylib             0x00007fffe33c5887 _pthread_body + 0
	10  libsystem_pthread.dylib             0x00007fffe33c508d thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6_

This seems to be due to the fact that you can't Poll from event outside the main thread, at least it seems to be an issue on MacOS, I don't have experience with SDL. Currently monitor.c initialises SDL and monitors for mouse and keyboard events.

To workaround this issue, what I did was.

I modified monitor_init() so that the SDL initialisation is done from main.c thread like so,

```
void monitor_init(void)
{
	/*Initialize the SDL*/
    SDL_Init(SDL_INIT_VIDEO);

    SDL_SetEventFilter(quit_filter, NULL);

	window = SDL_CreateWindow("TFT Simulator",
		SDL_WINDOWPOS_UNDEFINED, SDL_WINDOWPOS_UNDEFINED,
		MONITOR_HOR_RES, MONITOR_VER_RES, 0);       /*last param. SDL_WINDOW_BORDERLESS to hide borders*/

	renderer = SDL_CreateRenderer(window, -1, 0);
	texture = SDL_CreateTexture(renderer,
		SDL_PIXELFORMAT_ARGB8888, SDL_TEXTUREACCESS_STATIC, MONITOR_HOR_RES, MONITOR_VER_RES);

	/*Initialize the frame buffer to gray (77 is an empirical value) */
	memset(tft_fb, 77, MONITOR_HOR_RES * MONITOR_VER_RES * sizeof(uint32_t));
	SDL_UpdateTexture(texture, NULL, tft_fb, MONITOR_HOR_RES * sizeof(uint32_t));
	sdl_refr_qry = true;

	SDL_CreateThread(sdl_refr, "sdl_refr", NULL);
	while(sdl_inited == false); /*Wait until 'sdl_refr' initializes the SDL*/
}
```

I then modified the monitor thread to just do the drawing routines,

```
static int sdl_refr(void * param)
{
    (void)param;
	sdl_inited = true;

	/*Run until quit event not arrives*/
	while(sdl_quit_qry == false) {

		/*Refresh handling*/
		if(sdl_refr_qry != false) {
            sdl_refr_qry = false;
            SDL_UpdateTexture(texture, NULL, tft_fb, MONITOR_HOR_RES * sizeof(uint32_t));
            SDL_RenderClear(renderer);
            SDL_RenderCopy(renderer, texture, NULL, NULL);
            SDL_RenderPresent(renderer);
		}

		/*Sleep some time*/
		SDL_Delay(SDL_REFR_PERIOD);
	}

	SDL_DestroyTexture(texture);
	SDL_DestroyRenderer(renderer);
	SDL_DestroyWindow(window);
	SDL_Quit();

	exit(0);

	return 0;
}
```

On the main.c thread I included the polling for mouse and keyboard events like so,

```
   while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(1000);       /*Just to let the system breath*/

	    SDL_Event event;
	    while(SDL_PollEvent(&event)) {
#if USE_MOUSE != 0
            mouse_handler(&event);
#endif

#if USE_KEYBOARD
            keyboard_handler(&event);
#endif
	    }

    }
```

There is probably a better way to do this, but because I have zero experience with SDL I'm not sure of other solution may apply.

Cheers,
Martin

kisvegabor, 2018-04-03:
Hi,

Thank you for the reporting the issue and telling our solution. 

I implemented SDL_Iinit in a thread because on Windows drawing was possible on in that "context" when the init was called. So the init was required to move to the thread where the drawing happens.

I will immerse in it as I will have time. Until that time your solution surely will help to others on OSX.

fishermr, 2018-08-01:
I ran into the same problem with SDL.  I am using Eclipse Photon Release (4.8.0) and was having the exact issue you described.  Waiting for hardware and wanted to get started understanding this UX library.  
Much appreciate for the time you put into finding this thoughtful solution.

Calebe94, 2018-08-12:
It works perfectly on my MacOS 10.14. I owe you a beer!

BrendanSimon, 2018-08-19:
I've got the same issue.
Is there a patch or a fork/branch that I can access to get a a working copy of the source to compile ??
Thanks :)

prsolucoes, 2018-08-27:
Hi,

I have create two pull requests to make it work on OSX latest version. Tested locally:

https://github.com/littlevgl/pc_simulator/pull/21
https://github.com/littlevgl/lv_drivers/pull/16

Thanks.

bikeNomad, 2018-10-17:
I just tried building from the github master and running on my system (High Sierra, eclipse, sdl2 installed from Homebrew) and got the same crash. The only change I needed to make to get it to build was to remove the `#define` of `USE_FBDEV` in `lv_drv_conf.h`.

It looks like the SDL event pump is still running from the monitor thread.

[crash_report.txt](https://github.com/littlevgl/pc_simulator/files/2488491/crash_report.txt)

ripe909, 2018-10-21:
Yep, I see the same crash again.

kisvegabor, 2018-10-27:
Hi,
Thanks for the feedback!

I fixed it here: https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L299
and here: https://github.com/littlevgl/lv_drivers/blob/master/lv_drv_conf_templ.h#L147