lvgl/lv_port_pc_eclipse #14: Allow cross-compilation

aenchevich, 2018-04-19:
Please change the hard coded "CC = gcc" in the Makefile to  "CC ?= gcc". This my entire proposed change, with the above being the most important part:

diff --git a/Makefile b/Makefile
index 4d9120d..1f6a6df 100644
--- a/Makefile
+++ b/Makefile
@@ -1,12 +1,12 @@
 #
 # Makefile
 #
-CC = gcc
+CC ?= gcc
 CFLAGS = -Wall -Wshadow -Wundef -Wmaybe-uninitialized
 CFLAGS += -O3 -g3 -I./
 LDFLAGS += -lSDL2 -lm
 BIN = demo
-VPATH = 
+VPATH =
 
 LVGL_DIR = ${shell pwd}
 
@@ -85,12 +85,14 @@ OBJS = $(AOBJS) $(COBJS)
 all: clean default
 
 %.o: %.c
-       @$(CC)  $(CFLAGS) -c $< -o $@
        @echo "CC $<"
-    
+       @$(CC)  $(CFLAGS) -c $< -o $@
+
 default: $(AOBJS) $(COBJS) $(MAINOBJ)
        $(CC) -o $(BIN) $(MAINOBJ) $(AOBJS) $(COBJS) $(LDFLAGS)
 
-clean: 
-       rm -f $(BIN) $(AOBJS) $(COBJS) $(MAINOBJ)
+clean:
+       rm -f $(AOBJS) $(COBJS) $(MAINOBJ)
 
+maintainer-clean:
+       rm -f $(BIN) $(AOBJS) $(COBJS) $(MAINOBJ)

kisvegabor, 2018-04-20:
Hi,

Thank you for the improvements!

Please add your changes i na form of pull request to better see what changed.
Thank you!

aenchevich, 2018-07-08:
This is fixed, thank you, closing :)

closed 2018-07-08 by aenchevich