lvgl/lv_port_pc_eclipse #20: Not working on OS X (High Sierra 10.13.6, Eclipse Oxygen)

BrendanSimon, 2018-08-13:
I've cloned pc_simulator, installed sdl with brew, imported the project into Eclipse Oxygen, and click the build icon.
`main.c` builds ok, but the link fails.

It seems that objects or libraries are missing.

I tried running `make` from the root directly, which then built the demo (which doesn't run).  I tried adding those objects to the pc_simulator link and it did compile, but again didn't run.

How can I get this running on OS X?

```
make all 
Building file: ../main.c
Invoking: Cross GCC Compiler
gcc -I"/Users/brendan/Sandbox/littlevgl/pc_simulator" -O0 -Og -g3 -Wall -c -fmessage-length=0 -MMD -MP -MF"main.d" -MT"main.o" -o "main.o" "../main.c"
Finished building: ../main.c
 
Building target: pc_simulator
Invoking: Cross GCC Linker
gcc  -o "pc_simulator"  ./main.o   
Undefined symbols for architecture x86_64:
  "_SDL_CreateThread", referenced from:
      _hal_init in main.o
  "_SDL_Delay", referenced from:
      _tick_thread in main.o
  "_demo_create", referenced from:
      _main in main.o
  "_lv_disp_drv_init", referenced from:
      _hal_init in main.o
  "_lv_disp_drv_register", referenced from:
      _hal_init in main.o
  "_lv_indev_drv_init", referenced from:
      _hal_init in main.o
  "_lv_indev_drv_register", referenced from:
      _hal_init in main.o
  "_lv_init", referenced from:
      _main in main.o
  "_lv_task_handler", referenced from:
      _main in main.o
  "_lv_tick_inc", referenced from:
      _tick_thread in main.o
  "_monitor_fill", referenced from:
      _hal_init in main.o
  "_monitor_flush", referenced from:
      _hal_init in main.o
  "_monitor_init", referenced from:
      _hal_init in main.o
  "_monitor_map", referenced from:
      _hal_init in main.o
  "_mouse_init", referenced from:
      _hal_init in main.o
  "_mouse_read", referenced from:
      _hal_init in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [pc_simulator] Error 1
```

kisvegabor, 2018-08-13:
Hi,

I'm don't use OSX but from "Undefined symbols for architecture x86_64:" it seems you have installed/linked 32 bit SDL to a 64 bit machine or the opposite.

BrendanSimon, 2018-08-13:
Note: I have built or installed `littlevgl/lvgl`.  I'm assuming that the `littlevgl/pc_simulator` is standalone and will build the necessary `lvgl` dependencies via git submodules.

Is that correct?  or do I need to make and install `littlevgl/lvgl` seperately?

kisvegabor, 2018-08-15:
> I'm assuming that the littlevgl/pc_simulator is standalone and will build the necessary lvgl dependencies via git submodules.

Yes, just build in Eclipse and it should work.

Pleas ask your question in this thread: https://github.com/littlevgl/pc_simulator/issues/12
Here some guys said they were able to build and run LittelvGL on OSX.

prsolucoes, 2018-08-27:
Hi,

I have create two pull requests to make it work on OSX latest version. Tested locally:

https://github.com/littlevgl/pc_simulator/pull/21
https://github.com/littlevgl/lv_drivers/pull/16

Thanks.