lvgl/lv_port_pc_eclipse #22: Window sometimes doesn't update

poelstra, 2018-09-26:
The simulator is a very nice tool!

However, sometimes the (SDL) window stops updating. I'm not sure of a clear way of reproducing yet, it works fine most of the time.

When the problem occurs, the window isn't redrawn, e.g. a blinking icon stops blinking, when I turn the mousewheel (encoder), the actual focus changes, but it isn't drawn on the screen.
Only in certain cases the window still does redraw, e.g. when I move my mouse over the minimize/close buttons of the SDL window (i.e. the OS title bar). So if I keep moving the mouse there, the window updates fine, as soon as I stop, the redrawing stops too.

I have to close the application and start it again. Sometimes it then doesn't work for a number of times, sometimes it starts working again. So far, it seems that it either 'just works' after starting and then keeps working, or it 'just doesn't work' after starting, and also stays broken until I restart.

I tried to debug it a bit, and it appears that `sdl_refr_qry` is still being set to `true` in `monitor_flush()` and being 'handled' (and set to `false`) in `monitor_sdl_refr_core()`. I compared the render code to SDL examples, and can't spot an obvious error in that either.

It happens both on my own code, and with `lv_test_group_1()`.

So I have no clue why it may be happening...

kisvegabor, 2018-09-26:
Hi,

Do you use the last version of `lv_drivers` `master`? There were some updates in this regards.

poelstra, 2018-09-26:
I was using one commit before the last one of lv_drivers, i.e. the one with my mouse wheel encoder merged in (i.e. your merged version).

kisvegabor, 2018-09-26:
I see. I never experienced in on Linux or Windows but I don't use OSX. It should be an SDL-OSX issue.

poelstra, 2018-09-26:
Forgot to mention, I'm on Ubuntu 14.04, with SDL 2.0.2.
Maybe it's actually an SDL issue, it's an old version after all.

It's workable enough for now, and it's a simulator anyway, so I can live with it :)

kisvegabor, 2018-09-26:
Strange, nobody reported anything similar so far. Probably it's just the old library. 
If it's usable for you and we have no idea how to solve it I close this issue. :(
Please comment here or add a pull request if you find a solution.

closed 2018-09-26 by kisvegabor

poelstra, 2018-09-27:
Will do, thanks.