lvgl/lv_port_pc_eclipse #23: Version for Codeblocks

Polarisru, 2018-09-28:
Hello,
I have just ported Eclipse project to Codevision, may be it would be interested for other users?
Codeblocks seems to be more quicky and reliable and doesn't need anything to be installed, MinGW comes already with the software and you need only SDL2 includes and libraries to be able to run simulation project.
Waiting for your reply.

kisvegabor, 2018-09-28:
Hello,

A CodeBlock project would very welcome. Thank you in advance!

For simplicity, this repository is only the Eclipse project but I would kindly add your project to https://github.com/littlevgl/lv_projects if you made a pull request.

ajithpv, 2018-10-05:
Since, CodeBlock is already added to https://github.com/littlevgl/lv_projects/tree/master/codeblocks, I think we can close this issue.

kisvegabor, 2018-10-05:
@ajithpv 
Agreed, thank you!

closed 2018-10-05 by kisvegabor