lvgl/lv_port_pc_eclipse #30: Build error on macOSX

ffzwadd, 2018-10-21:
lv_drivers/display/fbdev.c:17:10: fatal error: 'linux/fb.h' file not found

neilabdev, 2018-10-24:
In summary, when compiling on mac you need to: 
* #define USE_FBDEV 0  // THIS DISABLES USE OF LINUX FB

* Add #ifdef macro around block in monitor.c file

like the following:

```
diff --git a/lv_drivers b/lv_drivers
--- a/lv_drivers
+++ b/lv_drivers
@@ -1 +1 @@
-Subproject commit 1e10b91f92f427a7de2ba1a81fb8fec88d8b852f
+Subproject commit 1e10b91f92f427a7de2ba1a81fb8fec88d8b852f-dirty
diff --git a/lv_drv_conf.h b/lv_drv_conf.h
index 9dce4bd..174b818 100644
--- a/lv_drv_conf.h
+++ b/lv_drv_conf.h
@@ -143,7 +143,7 @@
 /*-----------------------------------------
  *  Linux frame buffer device (/dev/fbx)
  *-----------------------------------------*/
-#define USE_FBDEV           1
+#define USE_FBDEV           0
 #if USE_FBDEV != 0
 #define FBDEV_PATH          "/dev/fb0"
 #endif

```

AND

```
--- a/lv_drivers
+++ b/lv_drivers
@@ -1 +1 @@
-Subproject commit 1e10b91f92f427a7de2ba1a81fb8fec88d8b852f
+Subproject commit 1e10b91f92f427a7de2ba1a81fb8fec88d8b852f-dirty
diff --git a/lv_drv_conf.h b/lv_drv_conf.h
index 9dce4bd..174b818 100644
--- a/lv_drv_conf.h
+++ b/lv_drv_conf.h
@@ -143,7 +143,7 @@
 /*-----------------------------------------
  *  Linux frame buffer device (/dev/fbx)
  *-----------------------------------------*/
-#define USE_FBDEV           1
+#define USE_FBDEV           0
 #if USE_FBDEV != 0
 #define FBDEV_PATH          "/dev/fb0"
 #endif
[ghost@redux pc_simulator>find . -name monitor.c
./lv_drivers/display/monitor.c
[ghost@redux pc_simulator>cd lv_drivers/
[ghost@redux lv_drivers>git diff
diff --git a/display/monitor.c b/display/monitor.c
index 20d2b7e..46e67fd 100644
--- a/display/monitor.c
+++ b/display/monitor.c
@@ -296,7 +296,7 @@ static void monitor_sdl_refr_core(void)
         SDL_RenderCopy(renderer, texture, NULL, NULL);
         SDL_RenderPresent(renderer);
     }
-
+#ifndef MONITOR_APPLE
     SDL_Event event;
     while(SDL_PollEvent(&event)) {
 #if USE_MOUSE != 0
@@ -329,6 +329,7 @@ static void monitor_sdl_refr_core(void)
 
     /*Sleep some time*/
     SDL_Delay(SDL_REFR_PERIOD);
+#endif
 }
 
 #endif
```

SEE PR: https://github.com/littlevgl/pc_simulator/issues/29

kisvegabor, 2018-10-27:
Fixed, see : https://github.com/littlevgl/pc_simulator/issues/12#issuecomment-433598977

ffzwadd, 2018-10-29:
> Fixed, see : [#12 (comment)](https://github.com/littlevgl/pc_simulator/issues/12#issuecomment-433598977)

Thanks, this works. But failed to run the demo.
Please see the log below:
2018-10-29 17:20:23.728 demo[27003:4540970] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff314742fb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff57dffc76 objc_exception_throw + 48
	2   AppKit                              0x00007fff2f1a02cf -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 4167
	3   libSDL2-2.0.0.dylib                 0x00000001054201fb Cocoa_PumpEvents + 211
	4   libSDL2-2.0.0.dylib                 0x00000001053a4847 SDL_PumpEvents_REAL + 23
	5   libSDL2-2.0.0.dylib                 0x00000001053a48c2 SDL_WaitEventTimeout_REAL + 76
	6   demo                                0x00000001050bd35d monitor_sdl_refr_thread + 141
	7   libSDL2-2.0.0.dylib                 0x00000001053ce6d3 SDL_RunThread + 60
	8   libSDL2-2.0.0.dylib                 0x000000010542c668 RunThread + 9
	9   libsystem_pthread.dylib             0x00007fff58c786c1 _pthread_body + 340
	10  libsystem_pthread.dylib             0x00007fff58c7856d _pthread_body + 0
	11  libsystem_pthread.dylib             0x00007fff58c77c5d thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

kisvegabor, 2018-11-06:
> Seems both of 'main' and 'monitor_sdl_refr_core' handled the he SDL events.

It should be solved here:
https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L299

Have you updated lv_drivers?

ffzwadd, 2018-11-10:
> > Seems both of 'main' and 'monitor_sdl_refr_core' handled the he SDL events.
> 
> It should be solved here:
> https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L299
> 
> Have you updated lv_drivers?

It works, Thanks.

closed 2018-11-10 by ffzwadd