lvgl/lv_port_pc_eclipse #32: video tutorial for pc_simulator_v5_2

himsha, 2018-12-21:
hi all,
is there is any video tutorial for pc_simulator_v5_2? - software will definitely will help a lot.as i had tried but not able to install it

embeddedt, 2018-12-21:
https://youtu.be/ZzLdct2ymvg

himsha, 2018-12-22:
tried a lot but now able to run simulator, do you have any video with window 10 os

himsha, 2018-12-24:
ohh i had wrote wrong i am not able to run simulator

embeddedt, 2018-12-24:
@kisvegabor My experience with making complex Linux applications ('complex' meaning dependencies on other libraries) work on Windows has never been very straightforward. If it's a CLI application, Cygwin/MSYS2 are very straightforward to use, but using them for GUI applications requires additional installation and setup.

Maybe doing a native Windows port of LittlevGL is a better option? If you're interested I could look into doing this sometime in the future.

embeddedt, 2018-12-25:
@kisvegabor Yes.

kisvegabor, 2018-12-27:
@embeddedt Sounds great! I suppose some new drivers would be required instead of `monitor`, `mouse`, etc. Or the existing ones should be extended with `#define`s?

embeddedt, 2018-12-27:
@kisvegabor The Windows architecture is sufficiently different that I think new drivers would be best.

kisvegabor, 2018-12-27:
@embeddedt 
> The Windows architecture is probably best served by new drivers.

Okay. You know it better. :)

> I was thinking I would use Code::Blocks for the project, is that OK?

We already have [Code::Blocks project](https://github.com/littlevgl/lv_projects/tree/master/codeblocks). Can it be updated?  

However, as Code::Blocks is also cross-platform IDE (like Eclipse) maybe  [Visual Studio](https://github.com/littlevgl/lv_projects/tree/master/visual_studio_2017_sdl_x64) would be better. 

Or we can use the current Eclipse project too and initialize the Windows drivers using `#if WIN32` (or similar) The only issue is it won't compile if SDL library is added to the project but not installed. But maybe it's better to ask the user to remove the library (3 clicks) than maintaining more PC projects. 

What do you think?

embeddedt, 2018-12-27:
@kisvegabor I was trying to avoid Visual Studio because I didn't want to place any dependency on proprietary products for a native Windows demo.

I think I can update the existing Code::Blocks project.

kisvegabor, 2018-12-27:
@embeddedt 
> I think I can update the existing Code::Blocks project.

Ok, let's do this way!

embeddedt, 2018-12-27:
@kisvegabor I expect to have it finished within a few days. Could you add me as a collaborator to the `lv_drivers` and `lv_projects` repositories so I can push my changes to a branch without needing to fork?

kisvegabor, 2018-12-28:
@embeddedt I sent the invitations

embeddedt, 2018-12-28:
@kisvegabor Thanks! I accepted them!

himsha, 2019-01-21:
when I am trying to do the changes as per explained here [https://docs.littlevgl.com/#PC-simulator](url), other setting of eclipse get disturbed, after this i am not able to built my existing project that was successfully building before. If someone had tried simulator before with window, i really appreciate. It took a good time with diso board to check new changes again and again.

kisvegabor, 2019-01-21:
On Windows you can use @embeddedt's new projects: https://github.com/littlevgl/pc_simulator_win_codeblocks

It should work out of the box.

embeddedt, 2019-01-21:
@kisvegabor Would you like me to add an Eclipse project? It would be exactly the same code, just a different IDE.

kisvegabor, 2019-01-21:
@embeddedt I think it"s very easy to make the current SDL-Eclipse project work with the Windows drivers:
1. Delete SDL library linkage from the Project properties
2. Enable the Windows drivers
3. Init. the windows drivers.

I think it'd be enough to describe it in comments and in the documentation of the PC simulator than maintaining two projects. What do you think?

The new Code blocks project should be mentioned in the documentation too. I'll update it when v5.3 released.

embeddedt, 2019-01-21:
@kisvegabor It works if people find it straightforward. Maybe we could put `#ifdef WIN32`s in the code so they would only have to do step 1.

kisvegabor, 2019-01-22:
@embeddedt Agreed!

closed 2019-02-28 by stale[bot]