lvgl/lv_port_pc_eclipse #34: Beginner Question: How I change background image in demo ?

KaungZawHtet, 2019-01-18:
I found this code block in demo.c but I don't know where should i change it. Nth is directly calling the bubble_pattern.png

embeddedt, 2019-01-18:
Here's where the image is set:

https://github.com/littlevgl/lv_examples/blob/088c9db7680764c9a8c9fdd40d0854968c64bd73/lv_apps/demo/demo.c#L65

And here is where it is declared:

https://github.com/littlevgl/lv_examples/blob/088c9db7680764c9a8c9fdd40d0854968c64bd73/lv_apps/demo/demo.c#L46

Basically, you'll want to create a new image using the [image converter](https://littlevgl.com/image-to-c-array) and then include the generated C file in your project. The image converter has a full explanation of the steps.

embeddedt, 2019-01-18:
@KaungZawHtet 2 things:

1. Are you sure you added the C file to the CMake file?
2. Maybe it is called `img_background` instead of `background`. You can check the bottom of the generated C file (where an `lv_img_dsc_t` variable is declared) to see the exact name to use.

KaungZawHtet, 2019-01-18:
yes, forget to add the line into CMake file. Thx, sir

closed 2019-01-18 by KaungZawHtet