lvgl/lv_port_pc_eclipse #36: Can you provide a CMakeLists. txt?

3038922, 2019-02-10:
I started learning Linux today. But I would never build a project without IDE.
my os is ubuntu 18.04
···
cmake_minimum_required(VERSION 3.10)
project(lvgl)
set(CMAKE_C_STANDARD 11)#C11
set(CMAKE_CXX_STANDARD 17)#C17
set(CMAKE_CXX_STANDARD_REQUIRED ON)

INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR})
file(GLOB_RECURSE INCLUDES "lv_drivers/*.h" "lv_drivers/*.hpp" "lv_examples/*.h" "lv_examples/*.hpp" "lvgl/*.h" "lvgl/*.hpp")

file(GLOB_RECURSE SOURCES "lv_drivers/*.c" "lv_drivers/*.cpp" "lv_examples/*.c" "lv_examples/*.cpp" "lvgl/*.c" "lvgl/*.cpp")

ADD_DEFINITIONS (-O3 -g3 -I  -Wall -Wshadow -Wundef -Wmaybe-uninitialized -fbounds-check) 

SET(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin)
SET(LIBRARY_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/firmware) 

LINK_DIRECTORIES(${LIBRARY_OUTPUT_PATH})

add_executable(main ${SOURCES} ${INCLUDES})

add_custom_target (run COMMAND ${EXECUTABLE_OUTPUT_PATH}/main）
···

kisvegabor, 2019-02-11:
I don't understand. If you don't want to build without an IDE why do you need Cmake? 
Just import the project into Eclipse and hit the build button.

Anyway, on Linux you can simply use `make`.

For Cmake, have a look at this: https://github.com/littlevgl/lvgl/issues/732

3038922, 2019-02-11:
> I don't understand. If you don't want to build without an IDE why do you need Cmake?
> Just import the project into Eclipse and hit the build button.
> 
> Anyway, on Linux you can simply use `make`.
> 
> For Cmake, have a look at this: [littlevgl/lvgl#732](https://github.com/littlevgl/lvgl/issues/732)

You are really the most patient and careful person I have ever met.
The main contest environment for me is that vscode + makefile cannot use ide. The default lvgl for contest environment is V5.1. I want to recompile lvglv5.3 into a static library and link it with cmake instead of the old version.

closed 2019-02-11 by 3038922