lvgl/lv_port_pc_eclipse #37: Run pc_simulator with WSL + XServer from Windows 10

beibean, 2019-02-25:
Hi,

I just wanted to share with you a new way to run simulation within Windows OS, with no need of an IDE.

All we need is:

1. Windows Subsystem for Linux (WSL from now on). You can check how to install it from [here](https://docs.microsoft.com/en-us/windows/wsl/install-win10)

2. Install a program to launch Server X within windows, eg: [VcXsrv](https://sourceforge.net/projects/vcxsrv/)

3. From WSL install SDL 2 dependencies for Linux as in [documentation](https://docs.littlevgl.com/#PC-simulator).

4. Run xLaunch application installed on step 2 and keep it open.

5. From WSL, compile project using: `$> make`.

6. After compiling finishes, launch demo from WSL: `$> ./demo`. Now, on xLaunch application we can see the simulated screen.

**Note:**some other dependencies may be required as indicated on [Dockerfile](https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/Dockerfile).

kisvegabor, 2019-02-25:
Hi, 
thank you for sharing this!

Would you be interested in writing a blog post about it on [LittlevGL's blog](https://blog.littlevgl.com/)?

Here is the [source of the blog](https://github.com/littlevgl/blog) and how to write a post.

beibean, 2019-02-25:
Hi,

I will add it to the blog yeah ;)

kisvegabor, 2019-02-25:
Awesome! :)

closed 2019-03-18 by kisvegabor