lvgl/lv_port_pc_eclipse #41: docker run lvgl_simulator block

JianwenLi-GS, 2019-04-29:
sudo docker run lvgl_simulator   **//block here until press CTRL+C**

^Cused:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624
used:  10824 ( 17 %), frag:   1 %, biggest free:  54624

embeddedt, 2019-04-29:
What's the problem here?

JianwenLi-GS, 2019-04-30:
How to use  lvgl_simulator docker? No output or graphical menu appears after running “sudo docker run lvgl_simulator ”

kisvegabor, 2019-04-30:
@BrianPugh, can you have a look at it? (The Dockerfile was added by Brian)

JianwenLi-GS, 2019-04-30:
> How to use lvgl_simulator docker? No output or graphical menu appears after running “sudo docker run lvgl_simulator ”

My PC‘s system is Ubuntu 18.04

BrianPugh, 2019-05-01:
see #42 . However, none of the Dockerfile changes actually impacts you (except maybe the instructions). This is a general Docker X11 forwarding issue. I added some brief instructions on how to (unsafely) X11 forward from a Docker container to your desktop environment. For more information, search "Docker X11 forwarding"

closed 2019-05-29 by stale[bot]