lvgl/lv_port_pc_eclipse #43: Description	Resource	Path	Location	Type Unknown referenced nature: org.python.pydev.pythonNature.	.project	/pc_simulator_sdl_eclipse	Unknown	Unknown nature

epikao, 2019-05-05:
Hello, just wanted to build an run PC_simulator 5.3, unfortunately with follow error,
what can be wrong here?:

Description	Resource	Path	Location	Type
Unknown referenced nature: org.python.pydev.pythonNature.	.project	/pc_simulator_sdl_eclipse	Unknown	Unknown nature

embeddedt, 2019-05-06:
@epikao If you're on Windows, I would recommend using CodeBlocks instead. It's designed to link to a native Windows application instead of depending on MinGW.

epikao, 2019-05-06:
ok. thanks, but which CodeBlocks Version is required? see below?
And do I need to install something more? Unfortunately there is no note in the readme file:
https://github.com/littlevgl/pc_simulator_win_codeblocks

codeblocks-17.12-setup.exe

codeblocks-17.12-setup-nonadmin.exe

codeblocks-17.12-nosetup.zip

codeblocks-17.12mingw-setup.exe

codeblocks-17.12mingw-nosetup.zip

codeblocks-17.12mingw_fortran-setup.exe

epikao, 2019-05-06:
Project/Target: "LittlevGL - Debug":
  The compiler's setup (GNU GCC Compiler) is invalid, so Code::Blocks cannot find/run the compiler.
  Probably the toolchain path within the compiler options is not setup correctly?!
  Do you have a compiler installed?
Goto "Settings->Compiler...->Global compiler settings->GNU GCC Compiler->Toolchain executables" and fix the compiler's setup.

Tried to run compiler executable 'C:\MinGW/bin/mingw32-gcc.exe', but failed!

so what now? I was thinking I don't need mingw for Codeblocks?

embeddedt, 2019-05-06:
@epikao Sorry, I never added this to the README file. You need the mingw version.

epikao, 2019-05-06:
so ok, but now I have 14 errors:
....codeblocks\pc_simulator_win_codeblocks-master\main.c|14|fatal error: lv_drivers/win_drv.h: No such file or directory|

embeddedt, 2019-05-06:
Did you download the sub modules?

epikao, 2019-05-06:
It works now with codeblocks, but after first run the demo works, but after close everthing crash and I need to reboot the computer...
So I'm back now to eclipse... now it compiles everthing without errors (only 5 warning) but I cannot run anything:

Program file does not exist
Debug\pc_simulator not found
Debug\pc_simulator not found
Debug\pc_simulator not found

embeddedt, 2019-05-07:
The `pc_simulator_sdl_eclipse` project seems to be designed for Eclipse on Linux. I highly recommend sticking with CodeBlocks. If you really want to use Eclipse on Windows you'll probably need to create a new Run configuration and point it to `Debug\pc_simulator.exe`.

> After first run the demo works properly but after close the demo something crashs and I need to reboot the computer

I doubt that LittlevGL is causing your whole computer to crash. Exactly what happens?

epikao, 2019-05-07:
> I doubt that LittlevGL is causing your whole computer to crash. Exactly what happens?

- open codeblocks
- open programm file
- build all
- run, start demo
- demo runs properly
- close demo with x because want change something on the code
- demo is closing, but it looks not properly because build button on codeblock is still not avaiable and the program can also not be deleted in the folder (because still someting in use).
- new start of computer
- now I can change code and build all, start demo without problem... but if want change again something not possible, if the demo already started. So need again to restart computer..

I can remember that pc_simulator V5.1 works properly with eclipse :-/ hmmm

embeddedt, 2019-05-07:
@epikao Try using the stop button inside CodeBlocks instead of the X on the title bar, or vice versa. There have been reports that only one of those works. I haven't found the root cause yet.

embeddedt, 2019-05-07:
@epikao And another option is to manually terminate LittlevGL.exe using Task Manager.

embeddedt, 2019-05-12:
@epikao Sorry, I tested it today and realized that it's the opposite of what I told you. The regular close button on the titlebar works fine. The "Stop" button inside CodeBlocks **won't** work.

closed 2019-06-09 by stale[bot]