lvgl/lv_port_pc_eclipse #45: Solved: git clone --recursive problem

ogunduz, 2019-07-10:
We try to clone with recursive but I got error

```sh
$ git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git
Cloning into 'pc_simulator_sdl_eclipse'...
remote: Enumerating objects: 30, done.
remote: Counting objects: 100% (30/30), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 936 (delta 13), reused 24 (delta 10), pack-reused 906
Receiving objects: 100% (936/936), 2.73 MiB | 67.00 KiB/s, done.
Resolving deltas: 100% (486/486), done.
Submodule 'lv_drivers' (https://github.com/littlevgl/lv_drivers.git) registered for path 'lv_drivers'
Submodule 'lv_examples' (https://github.com/littlevgl/lv_examples.git) registered for path 'lv_examples'
Submodule 'lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'lvgl'
Cloning into '/home/user/works/littlevgl/pc_simulator_sdl_eclipse/lv_drivers'...
remote: Enumerating objects: 90, done.        
remote: Counting objects: 100% (90/90), done.        
remote: Compressing objects: 100% (46/46), done.        
remote: Total 1040 (delta 51), reused 76 (delta 44), pack-reused 950        
Receiving objects: 100% (1040/1040), 269.13 KiB | 318.00 KiB/s, done.
Resolving deltas: 100% (709/709), done.
Cloning into '/home/user/works/littlevgl/pc_simulator_sdl_eclipse/lv_examples'...
remote: Enumerating objects: 251, done.        
remote: Counting objects: 100% (251/251), done.        
remote: Compressing objects: 100% (132/132), done.        
remote: Total 4738 (delta 145), reused 215 (delta 118), pack-reused 4487        
Receiving objects: 100% (4738/4738), 8.04 MiB | 244.00 KiB/s, done.
Resolving deltas: 100% (3286/3286), done.
Cloning into '/home/user/works/littlevgl/pc_simulator_sdl_eclipse/lvgl'...
remote: Enumerating objects: 87, done.        
remote: Counting objects: 100% (87/87), done.        
remote: Compressing objects: 100% (56/56), done.        
remote: Total 19701 (delta 48), reused 49 (delta 31), pack-reused 19614        
Receiving objects: 100% (19701/19701), 14.33 MiB | 245.00 KiB/s, done.
Resolving deltas: 100% (15621/15621), done.
error: Server does not allow request for unadvertised object 39fe8649543dcb2a2777ccee5d0b6714ee41f3bd
Fetched in submodule path 'lv_drivers', but it did not contain 39fe8649543dcb2a2777ccee5d0b6714ee41f3bd. Direct fetching of that commit failed.
```

And we found a workaround solution

```sh
$ cd pc_simulator_sdl_eclipse
$ rm -rf lv_drivers/ lv_examples/ lvgl/

$ git submodule update --remote
Submodule path 'lv_drivers': checked out 'b03533b909d1ec569700a72879b1ec7b4fb7c0a9'
Submodule path 'lv_examples': checked out '9a5a95ac4d3b86b62cb0917d3afdebb85bc990d3'
Submodule path 'lvgl': checked out '2cead48cd5ee23de05f7b11244021bb91f97d5e7
```

## :clipboard: Reference
1. https://stackoverflow.com/questions/1979167/git-submodule-update

embeddedt, 2019-07-10:
Yep, looks like we are referencing a nonexistent commit in the submodule. I'll fix it today.

embeddedt, 2019-07-11:
@kisvegabor should have fixed it in #46, does it work for you now?

closed 2019-08-08 by stale[bot]