lvgl/lv_port_pc_eclipse #47: compile errors on Windows

syeg, 2019-08-25:
Trying to compile PC Simulator on Windows 10 after following  instruction on https://docs.littlevgl.com/en/html/get-started/pc-simulator.html

BUT ...got compile error due to the fact that lv_drivers/display/fbdev.c has  include statements such as

#include <linux/fb.h>
#include <sys/mman.h>
#include <sys/ioctl.h>

Where do you download the include files?

Thx.

-------------
18:21:08 **** Incremental Build of configuration Debug for project pc_simulator ****
mingw32-make all 
'Building file: ../lv_drivers/display/fbdev.c'
'Invoking: Cross GCC Compiler'
gcc -I"D:\Eclipse_Cpp\pc_simulator_sdl_eclipse" -I"D:\Eclipse_Cpp\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0\mingw32\bin" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -W -MMD -MP -MF"lv_drivers/display/fbdev.d" -MT"lv_drivers/display/fbdev.o" -o "lv_drivers/display/fbdev.o" "../lv_drivers/display/fbdev.c"
../lv_drivers/display/fbdev.c:17:10: fatal error: linux/fb.h: No such file or directory
 #include <linux/fb.h>
          ^~~~~~~~~~~~
compilation terminated.
mingw32-make: *** [lv_drivers/display/subdir.mk:35: lv_drivers/display/fbdev.o] Error 1
"mingw32-make all" terminated with exit code 2. Build might be incomplete.

---/**
 * @file fbdev.c
 *
 */

/*********************
 *      INCLUDES
 *********************/
#include "fbdev.h"
#if USE_FBDEV

#include <stdlib.h>
#include <unistd.h>
#include <stddef.h>
#include <stdio.h>
#include <fcntl.h>
#include <linux/fb.h>
#include <sys/mman.h>
#include <sys/ioctl.h>
-

embeddedt, 2019-08-25:
It looks like we are enabling `USE_FBDEV` for some reason. That is causing the compilation errors.

As a workaround, change [this](https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/5e1039be56ab933c19aa98447bbd119062ec1e5a/lv_drv_conf.h#L186) to `0` instead of `1`. That should fix the issue.

syeg, 2019-08-25:
Thanks for the tip. I tried the fix nut encountered new errors:
16:23:42 **** Build of configuration Debug for project pc_simulator ****
mingw32-make all 
'Building target: pc_simulator'
'Invoking: Cross GCC Linker'
gcc -L"D:\Eclipse_Cpp\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0\mingw32\lib\gcc\i686-w64-mingw32\lib" -o "pc_simulator"  ./lvgl/src/lv_themes/lv_theme.o ./lvgl/src/lv_themes/lv_theme_alien.o ./lvgl/src/lv_themes/lv_theme_default.o ./lvgl/src/lv_themes/lv_theme_material.o ./lvgl/src/lv_themes/lv_theme_mono.o ./lvgl/src/lv_themes/lv_theme_nemo.o ./lvgl/src/lv_themes/lv_theme_night.o ./lvgl/src/lv_themes/lv_theme_templ.o ./lvgl/src/lv_themes/lv_theme_zen.o  ./lvgl/src/lv_objx/lv_arc.o ./lvgl/src/lv_objx/lv_bar.o ./lvgl/src/lv_objx/lv_btn.o ./lvgl/src/lv_objx/lv_btnm.o ./lvgl/src/lv_objx/lv_calendar.o ./lvgl/src/lv_objx/lv_canvas.o ./lvgl/src/lv_objx/lv_cb.o ./lvgl/src/lv_objx/lv_chart.o ./lvgl/src/lv_objx/lv_cont.o ./lvgl/src/lv_objx/lv_ddlist.o ./lvgl/src/lv_objx/lv_gauge.o ./lvgl/src/lv_objx/lv_img.o ./lvgl/src/lv_objx/lv_imgbtn.o ./lvgl/src/lv_objx/lv_kb.o ./lvgl/src/lv_objx/lv_label.o ./lvgl/src/lv_objx/lv_led.o ./lvgl/src/lv_objx/lv_line.o ./lvgl/src/lv_objx/lv_list.o ./lvgl/src/lv_objx/lv_lmeter.o ./lvgl/src/lv_objx/lv_mbox.o ./lvgl/src/lv_objx/lv_objx_templ.o ./lvgl/src/lv_objx/lv_page.o ./lvgl/src/lv_objx/lv_preload.o ./lvgl/src/lv_objx/lv_roller.o ./lvgl/src/lv_objx/lv_slider.o ./lvgl/src/lv_objx/lv_spinbox.o ./lvgl/src/lv_objx/lv_sw.o ./lvgl/src/lv_objx/lv_ta.o ./lvgl/src/lv_objx/lv_table.o ./lvgl/src/lv_objx/lv_tabview.o ./lvgl/src/lv_objx/lv_tileview.o ./lvgl/src/lv_objx/lv_win.o  ./lvgl/src/lv_misc/lv_anim.o ./lvgl/src/lv_misc/lv_area.o ./lvgl/src/lv_misc/lv_async.o ./lvgl/src/lv_misc/lv_circ.o ./lvgl/src/lv_misc/lv_color.o ./lvgl/src/lv_misc/lv_fs.o ./lvgl/src/lv_misc/lv_gc.o ./lvgl/src/lv_misc/lv_ll.o ./lvgl/src/lv_misc/lv_log.o ./lvgl/src/lv_misc/lv_math.o ./lvgl/src/lv_misc/lv_mem.o ./lvgl/src/lv_misc/lv_task.o ./lvgl/src/lv_misc/lv_templ.o ./lvgl/src/lv_misc/lv_txt.o ./lvgl/src/lv_misc/lv_utils.o  ./lvgl/src/lv_hal/lv_hal_disp.o ./lvgl/src/lv_hal/lv_hal_indev.o ./lvgl/src/lv_hal/lv_hal_tick.o  ./lvgl/src/lv_font/lv_font.o ./lvgl/src/lv_font/lv_font_fmt_txt.o ./lvgl/src/lv_font/lv_font_roboto_12.o ./lvgl/src/lv_font/lv_font_roboto_16.o ./lvgl/src/lv_font/lv_font_roboto_22.o ./lvgl/src/lv_font/lv_font_roboto_28.o ./lvgl/src/lv_font/lv_font_unscii_8.o  ./lvgl/src/lv_draw/lv_draw.o ./lvgl/src/lv_draw/lv_draw_arc.o ./lvgl/src/lv_draw/lv_draw_basic.o ./lvgl/src/lv_draw/lv_draw_img.o ./lvgl/src/lv_draw/lv_draw_label.o ./lvgl/src/lv_draw/lv_draw_line.o ./lvgl/src/lv_draw/lv_draw_rect.o ./lvgl/src/lv_draw/lv_draw_triangle.o ./lvgl/src/lv_draw/lv_img_cache.o ./lvgl/src/lv_draw/lv_img_decoder.o  ./lvgl/src/lv_core/lv_disp.o ./lvgl/src/lv_core/lv_group.o ./lvgl/src/lv_core/lv_indev.o ./lvgl/src/lv_core/lv_obj.o ./lvgl/src/lv_core/lv_refr.o ./lvgl/src/lv_core/lv_style.o  ./lvgl/porting/lv_port_disp_template.o ./lvgl/porting/lv_port_fs_template.o ./lvgl/porting/lv_port_indev_template.o  ./lv_examples/lv_tutorial/9_responsive/lv_tutorial_responsive.o  ./lv_examples/lv_tutorial/8_animations/lv_tutorial_animations.o  ./lv_examples/lv_tutorial/7_fonts/arial_20.o ./lv_examples/lv_tutorial/7_fonts/lv_tutorial_fonts.o  ./lv_examples/lv_tutorial/6_images/flower_icon_alpha.o ./lv_examples/lv_tutorial/6_images/lv_tutorial_images.o ./lv_examples/lv_tutorial/6_images/red_flower.o ./lv_examples/lv_tutorial/6_images/red_rose_16.o  ./lv_examples/lv_tutorial/5_antialiasing/apple_icon_alpha.o ./lv_examples/lv_tutorial/5_antialiasing/apple_icon_chroma.o ./lv_examples/lv_tutorial/5_antialiasing/lv_tutorial_antialiasing.o  ./lv_examples/lv_tutorial/4_themes/lv_tutorial_themes.o  ./lv_examples/lv_tutorial/3_styles/lv_tutorial_styles.o  ./lv_examples/lv_tutorial/2_objects/lv_tutorial_objects.o  ./lv_examples/lv_tutorial/1_hello_world/lv_tutorial_hello_world.o  ./lv_examples/lv_tutorial/10_keyboard/lv_tutorial_keyboard.o  ./lv_examples/lv_tests/lv_test_theme/lv_test_theme_1.o ./lv_examples/lv_tests/lv_test_theme/lv_test_theme_2.o  ./lv_examples/lv_tests/lv_test_stress/lv_test_stress.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_win/lv_test_win.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_tileview/lv_test_tileview.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_tabview/lv_test_tabview.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_table/lv_test_table.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_ta/lv_test_ta.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_sw/lv_test_sw.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_slider/lv_test_slider.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_roller/lv_test_roller.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_preload/lv_test_preload.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_page/lv_test_page.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_mbox/lv_test_mbox.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_lmeter/lv_test_lmeter.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_list/lv_test_list.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_line/lv_test_line.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_led/lv_test_led.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_label/lv_test_label.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_kb/lv_test_kb.o  ./lv_examples/lv_tests/lv_test_objx/lv_test_imgbtn/imgbtn_img_1.o ./lv_examples/lv_tests/lv_test_objx/lv_test_imgbtn/imgbtn_img_2.o ./lv_examples/lv_tests/lv_test_objx/lv_test_imgbtn/imgbtn_img_3.o ./lv_examples/lv_tests [... cut ...]
./main.o: In function `hal_init':
D:\Eclipse_Cpp\pc_simulator_sdl_eclipse\Debug/../main.c:174: undefined reference to `SDL_CreateThread'
./main.o: In function `tick_thread':
D:\Eclipse_Cpp\pc_simulator_sdl_eclipse\Debug/../main.c:191: undefined reference to `SDL_Delay'
collect2.exe: error: ld returned 1 exit status
mingw32-make: *** [makefile:91: pc_simulator] Error 1
"mingw32-make all" terminated with exit code 2. Build might be incomplete.

16:23:52 Build Failed. 20 errors, 0 warnings. (took 10s.8ms)

embeddedt, 2019-08-26:
You must have changed something else, as changing that setting alone shouldn't affect the build.

If you are on Windows, I suggest using the [CodeBlocks project](https://github.com/littlevgl/pc_simulator_win_codeblocks) instead.

kisvegabor, 2019-08-26:
Or on Windows, you can use the [Visual Studio](https://github.com/littlevgl/pc_simulator_sdl_visual_studio) project too. Similarly to the Code blocks project it also works out of the box without further configuration.

4lexChapman, 2019-09-19:
@syeg The undefined reference errors are likely due to not installing the SDL2 headers/libraries correctly. Did you follow the directions [here](https://docs.littlevgl.com/en/html/get-started/pc-simulator.html#install-sdl-2)?

syeg, 2019-09-20:
Yes....it's been a while now...I did it for Windows (10)  and after the instillation, there were no SDL2 related libs for compile so I have to downloaded it from somewhere and copied over the SDL2 libs to move ahead. Then I hit this compile error.

Then I moved to use CodeBlocks project as suggested and it did not compile successfully.

Then I tried it with Visual Studio and still no able to pass initial compile.

By this time I have spent enough time on this so I decide not to forge ahead...would be nice to have it fixed for WIndows.

Also would it work for WSL (Windows subsystem for Linux)?

Thanks a lot for the replies.

embeddedt, 2019-09-20:
>  use CodeBlocks project as suggested and it did not compile successfully.

It should work. Can you explain what didn't work?

closed 2019-10-18 by stale[bot]