lvgl/lv_port_pc_eclipse #49: SegFault in Demo

4lexChapman, 2019-09-19:
During demo_create()->write_create(), .... Every call to get_glyph_dsc_id() (in lv_font_fmt_txt.c), when trying to assign to fdsc members (last_letter or last_glyph_id), I get a segfault. Fortunately, these are just caching assignments, and commenting out these assignments (there are 4 in total) allows the code to run, but I would like to help others who might run into this. 

From digging around the code, it would seem that the pointer font->dsc exists and there is memory at the pointer. Other members at that pointer seem to have valid data, too. However, there seems to be a problem writing to the "last" members at this memory. I just did a test assignment to another part of that struct (cmap_num) and it also segfaulted.

Does anyone have any ideas?

Background: I downloaded (not cloned) the PC simulator code. Had to change #define USE_FBDEV to 0. I am using Eclipse CDT and MinGW64. No other changes.

embeddedt, 2019-09-19:
Interesting. I wonder if it's an issue that is being revealed by the MinGW compiler, since I have yet to experience the problem here on Linux (although, I haven't pulled the latest simulator in a while).

Can you try going into the `lvgl` and `lv_examples` directories, running `git checkout master` followed by `git pull`, and seeing if that fixes the issue?

4lexChapman, 2019-09-19:
Thanks @embeddedt for your quick reply. As I mentioned, I had downloaded the code, instead of checking out the repo. I had a feeling that would bite me.

After your suggestion, I cloned the repo to a new folder and it worked without incident. I did a quick diff between the repo files and the zipped files and found several differences, so perhaps the zipped version should be refreshed.

closed 2019-09-19 by 4lexChapman

embeddedt, 2019-09-19:
@4lexChapman It's because the cloned version will receive updates faster than the zip file, since the zip file has to be manually regenerated using a script (while the cloned version gets updates "as they happen").

The easiest way to work around that is to just manually run updates on the three directories inside (`lvgl`, `lv_examples`, and `lv_drivers`), which is a good idea to do even with the cloned version as we don't immediately update the submodules on `master`.

embeddedt, 2019-09-19:
> perhaps the zipped version should be refreshed.

Probably.

@kisvegabor Would you mind updating it when you get a chance?

kisvegabor, 2019-09-20:
> @kisvegabor Would you mind updating it when you get a chance?

Updated.