lvgl/lv_port_pc_eclipse #51: Mac Linking problem with pc simulator in Catalina

querist, 2019-10-19:
This problem was reportedly addressed previously, but with the latest code I'm running into the linking problem again trying to build the simulator.

12:21:34 **** Incremental Build of configuration Debug for project pc_simulator ****
make all 
Building target: pc_simulator
Invoking: Cross GCC Linker
gcc  -o "pc_simulator"  ./main.o ./mouse_cursor_icon.o   -lSDL2main -lSDL2 -lm
Undefined symbols for architecture x86_64:
  "_demo_create", referenced from:
      _main in main.o
  "_keyboard_handler", referenced from:
      _main in main.o
  "_lv_disp_buf_init", referenced from:
      _hal_init in main.o
  "_lv_disp_drv_init", referenced from:
      _hal_init in main.o
  "_lv_disp_drv_register", referenced from:
      _hal_init in main.o
  "_lv_disp_get_scr_act", referenced from:
      _hal_init in main.o
  "_lv_img_create", referenced from:
      _hal_init in main.o
  "_lv_img_set_src", referenced from:
      _hal_init in main.o
  "_lv_img_src_get_type", referenced from:
      _ICON_LVGL_runLength_info in main.o
  "_lv_indev_drv_init", referenced from:
      _hal_init in main.o
  "_lv_indev_drv_register", referenced from:
      _hal_init in main.o
  "_lv_indev_set_cursor", referenced from:
      _hal_init in main.o
  "_lv_init", referenced from:
      _main in main.o
  "_lv_mem_monitor", referenced from:
      _memory_monitor in main.o
  "_lv_task_create", referenced from:
      _hal_init in main.o
  "_lv_task_handler", referenced from:
      _main in main.o
  "_lv_tick_inc", referenced from:
      _tick_thread in main.o
  "_monitor_flush", referenced from:
      _hal_init in main.o
  "_monitor_init", referenced from:
      _hal_init in main.o
  "_mouse_handler", referenced from:
      _main in main.o
  "_mouse_init", referenced from:
      _hal_init in main.o
  "_mouse_read", referenced from:
      _hal_init in main.o
  "_mousewheel_handler", referenced from:
      _main in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [pc_simulator] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

I have the SDL2 libraries from brew.

embeddedt, 2019-10-19:
It appears that LittlevGL is not being linked into your project (but SDL2 is).

> gcc -o "pc_simulator" ./main.o ./mouse_cursor_icon.o -lSDL2main -lSDL2 -lm

This should have a lot of `lv_xxx.o` files as well.

querist, 2019-10-19:
> It appears that LittlevGL is not being linked into your project (but SDL2 is).
> 
> > gcc -o "pc_simulator" ./main.o ./mouse_cursor_icon.o -lSDL2main -lSDL2 -lm
> 
> This should have a lot of `lv_xxx.o` files as well.

This is the project that was from the website. I have not modified the code other than the screen size and colour depth in one header file. I'll see what I can find to fix that. If you have any ideas, I'd appreciate the info. Thanks.

querist, 2019-10-19:
New problem - I added the files back into the project using the import feature this time instead of just copying them and now I have this error:

../lv_drivers/display/fbdev.c:26:10: fatal error: 'linux/fb.h' file not found
#include <linux/fb.h>

I'm on a Mac. Why is it looking for linux files?
Also, I tried defining USE_BSD_FBDEV and that caused its own problems.

embeddedt, 2019-10-19:
You shouldn't need to copy the files into the project. If you download the project, unzip it, and import the whole project into Eclipse, things should work. Manually importing or copying certain files is extremely likely to cause problems.

querist, 2019-10-19:
The lvgl and related directories were empty and it would not compile, so I
copied the files and it compiled.

embeddedt, 2019-10-19:
> The lvgl and related directories were empty and it would not compile.

You must not have downloaded the repository correctly. Can you clone this repository using `git clone --recursive` and try it?

querist, 2019-10-19:
Sure. I’m in the grocery store at the moment so it will be a bit before I
can do that. Thanks.

querist, 2019-10-19:
> > The lvgl and related directories were empty and it would not compile.
> 
> You must not have downloaded the repository correctly. Can you clone this repository using `git clone --recursive` and try it?

I did the git clone --recursive and I'm receiving the same error:
../lv_drivers/display/fbdev.c:26:10: fatal error: 'linux/fb.h' file not found
#include <linux/fb.h>
         ^~~~~~~~~~~~
1 error generated.
make: *** [lv_drivers/display/fbdev.o] Error 1

Again, I'm on a Mac so I don't see why I should be using a linux file when there is apparently a BSD option available, but the BSD option causes its own problems with missing files:

../lv_drivers/display/fbdev.c:23:10: fatal error: 'sys/consio.h' file not found
#include <sys/consio.h>
         ^~~~~~~~~~~~~~
1 error generated.
make: *** [lv_drivers/display/fbdev.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

embeddedt, 2019-10-19:
Your comment sparked my memory.

Change [this define](https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/4c185d3f2b0d0a09ea0858e516697763c016976e/lv_drv_conf.h#L186) to a `0` in `lv_drv_conf.h` and try again.

embeddedt, 2019-10-19:
As far as I recall, someone else reported this issue before but I guess neither @kisvegabor nor I ever fixed the file.

querist, 2019-10-19:
That did it! Thanks.

kisvegabor, 2019-10-21:
I fixed it. Thanks!
525e33d

closed 2019-10-21 by kisvegabor