lvgl/lv_port_pc_eclipse #56: Eclipse simulator project: Release build error (arial20.c)

XFer012, 2020-04-20:
Hello, 
building the LittleVL pc_simulator Eclipse project (Eclipse IDE for C/C++ Developers Version: 2020-03 (4.15.0) Build id: 20200313-1211) on Ubuntu 18.04 fails to build in Release configuration.
Builds OK in Debug configuration.

Build error:
```

Building file: ../lv_examples/lv_tutorial/7_fonts/arial_20.c
Invoking: Cross GCC Compiler
gcc -I"/home/fernando/src/eclipse-workspace" -O3 -Wall -c -fmessage-length=0 -MMD -MP -MF"lv_examples/lv_tutorial/7_fonts/arial_20.d" -MT"lv_examples/lv_tutorial/7_fonts/arial_20.o" -o "lv_examples/lv_tutorial/7_fonts/arial_20.o" "../lv_examples/lv_tutorial/7_fonts/arial_20.c"
../lv_examples/lv_tutorial/7_fonts/arial_20.c:1:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [lv_examples/lv_tutorial/7_fonts/arial_20.o] Error 1
lv_examples/lv_tutorial/7_fonts/subdir.mk:21: recipe for target 'lv_examples/lv_tutorial/7_fonts/arial_20.o' failed
"make all" terminated with exit code 2. Build might be incomplete.

```

embeddedt, 2020-04-20:
I'm not sure if building in Release mode is supported with this project.

XFer012, 2020-04-23:
Not sure if it's the same issue: I tried building the demo con my embedded Linux system, running Ubuntu 18.04, using the included Makefile.

I get an error when compiling fbdev.c (after lots of modules correctly compiled):

`
PC_simulator_orig/lv_drivers/display/fbdev.c:9:

/root/src/LittleVGL/MyDemo02/PC_simulator_orig/lv_drivers/display/fbdev.h:29:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
Makefile:35: recipe for target 'fbdev.o' failed
make: *** [fbdev.o] Error 1
`

closed 2020-05-22 by stale[bot]