lvgl/lv_port_pc_eclipse #58: macOS build simulator pass, but not dispaly

zhepoch, 2020-05-21:


zhepoch, 2020-05-21:
Sample of demo:
```
Sampling process 35455 for 3 seconds with 1 millisecond of run time between samples
Sampling completed, processing symbols...
Analysis of sampling demo (pid 35455) every 1 millisecond
Process:         demo [35455]
Path:            /Users/USER/*/demo
Load Address:    0x10badf000
Identifier:      demo
Version:         0
Code Type:       X86-64
Parent Process:  zsh [31155]

Date/Time:       2020-05-21 17:54:08.126 +0800
Launch Time:     2020-05-21 17:51:50.133 +0800
OS Version:      Mac OS X 10.14.6 (18G103)
Report Version:  7
Analysis Tool:   /usr/bin/sample

Physical footprint:         31.4M
Physical footprint (peak):  31.6M
----

Call graph:
    2613 Thread_354651   DispatchQueue_1: com.apple.main-thread  (serial)
    + 2613 start  (in libdyld.dylib) + 1  [0x7fff6f51d3d5]
    +   2564 main  (in demo) + 271  [0x10badfb2f]
    +   ! 2563 usleep  (in libsystem_c.dylib) + 53  [0x7fff6f5e1808]
    +   ! : 2560 nanosleep  (in libsystem_c.dylib) + 199  [0x7fff6f5e1914]
    +   ! : | 2531 __semwait_signal  (in libsystem_kernel.dylib) + 10,12  [0x7fff6f655f32,0x7fff6f655f34]
    +   ! : | 16 cerror  (in libsystem_kernel.dylib) + 20  [0x7fff6f652435]
    +   ! : | + 16 cerror_nocancel  (in libsystem_kernel.dylib) + 20,0,...  [0x7fff6f65246b,0x7fff6f652457,...]
    +   ! : | 12 cerror  (in libsystem_kernel.dylib) + 0,1  [0x7fff6f652421,0x7fff6f652422]
    +   ! : | 1 cerror  (in libsystem_kernel.dylib) + 13  [0x7fff6f65242e]
    +   ! : |   1 _pthread_exit_if_canceled  (in libsystem_pthread.dylib) + 0  [0x7fff6f70ea7c]
    +   ! : 2 nanosleep  (in libsystem_c.dylib) + 151  [0x7fff6f5e18e4]
    +   ! : 1 nanosleep  (in libsystem_c.dylib) + 208  [0x7fff6f5e191d]
    +   ! :   1 __error  (in libsystem_kernel.dylib) + 0  [0x7fff6f652b65]
    +   ! 1 usleep  (in libsystem_c.dylib) + 0  [0x7fff6f5e17d3]
    +   45 main  (in demo) + 261  [0x10badfb25]
    +   ! 41 lv_task_handler  (in demo) + 757  [0x10bb2e415]
    +   ! : 36 anim_task  (in demo) + 253  [0x10bb2f75d]
    +   ! : | 6 arc_anim  (in demo) + 78  [0x10bb52c8e]
    +   ! : | + 1 lv_label_refr_text  (in demo) + 179  [0x10bb14653]
    +   ! : | + ! 1 _lv_obj_get_style_ptr  (in demo) + 191  [0x10baed26f]
    +   ! : | + !   1 lv_obj_get_state  (in demo) + 53  [0x10baecd45]
    +   ! : | + !     1 lv_debug_check_obj_type  (in demo) + 32  [0x10baf4920]
    +   ! : | + !       1 lv_obj_get_type  (in demo) + 81  [0x10baedae1]
    +   ! : | + !         1 _lv_memset_00  (in demo) + 312  [0x10bb301b8]
    +   ! : | + !           1 _platform_bzero$VARIANT$Haswell  (in libsystem_platform.dylib) + 119  [0x7fff6f705997]
    +   ! : | + 1 lv_label_refr_text  (in demo) + 198  [0x10bb14666]
    +   ! : | + ! 1 _lv_obj_get_style_int  (in demo) + 277  [0x10baecc15]
    +   ! : | + !   1 lv_debug_check_obj_type  (in demo) + 32  [0x10baf4920]
    +   ! : | + !     1 lv_obj_get_type  (in demo) + 8  [0x10baeda98]
    +   ! : | + 1 lv_label_refr_text  (in demo) + 416  [0x10bb14740]
    +   ! : | + ! 1 lv_obj_set_size  (in demo) + 670  [0x10bae641e]
    +   ! : | + !   1 lv_label_signal  (in demo) + 658  [0x10bb13b62]
    +   ! : | + !     1 lv_label_refr_text  (in demo) + 416  [0x10bb14740]
    +   ! : | + !       1 lv_obj_set_size  (in demo) + 177  [0x10bae6231]
    +   ! : | + !         1 lv_debug_check_obj_valid  (in demo) + 83  [0x10baf4a53]
    +   ! : | + !           1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !             1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !               1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                 1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                   1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                     1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                       1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                         1 obj_valid_child  (in demo) + 67  [0x10baf4ad3]
    +   ! : | + !                           1 obj_valid_child  (in demo) + 82  [0x10baf4ae2]
    +   ! : | + !                             1 _lv_ll_get_next  (in demo) + 1  [0x10bb326d1]
[... 599 lines cut ...]
    0x7fff6f54c000 -     0x7fff6f54dff7  libremovefile.dylib (45.200.2) <9FBEB2FF-EEBE-31BC-BCFC-C71F8D0E99B6> /usr/lib/system/libremovefile.dylib
    0x7fff6f54e000 -     0x7fff6f565ff3  libsystem_asl.dylib (356.200.4) <A62A7249-38B8-33FA-9875-F1852590796C> /usr/lib/system/libsystem_asl.dylib
    0x7fff6f566000 -     0x7fff6f566ff7  libsystem_blocks.dylib (73) <A453E8EE-860D-3CED-B5DC-BE54E9DB4348> /usr/lib/system/libsystem_blocks.dylib
    0x7fff6f567000 -     0x7fff6f5eefff  libsystem_c.dylib (1272.250.1) <7EDACF78-2FA3-35B8-B051-D70475A35117> /usr/lib/system/libsystem_c.dylib
    0x7fff6f5ef000 -     0x7fff6f5f2ffb  libsystem_configuration.dylib (963.270.3) <2B4A836D-68A4-33E6-8D48-CD4486B03387> /usr/lib/system/libsystem_configuration.dylib
    0x7fff6f5f3000 -     0x7fff6f5f6ff7  libsystem_coreservices.dylib (66) <719F75A4-74C5-3BA6-A09E-0C5A3E5889D7> /usr/lib/system/libsystem_coreservices.dylib
    0x7fff6f5f7000 -     0x7fff6f5fdfff  libsystem_darwin.dylib (1272.250.1) <EC9B39A5-9592-3577-8997-7DC721D20D8C> /usr/lib/system/libsystem_darwin.dylib
    0x7fff6f5fe000 -     0x7fff6f604ff7  libsystem_dnssd.dylib (878.270.2) <E9A5ACCF-E35F-3909-AF0A-2A37CD217276> /usr/lib/system/libsystem_dnssd.dylib
[... cut ...]
    0x7fff6f720000 -     0x7fff6f727ff3  libsystem_symptoms.dylib (820.267.1) <03F1C2DD-0F5A-3D9D-88F6-B26C0F94EB52> /usr/lib/system/libsystem_symptoms.dylib
    0x7fff6f728000 -     0x7fff6f73dfff  libsystem_trace.dylib (906.260.1) <FC761C3B-5434-3A52-912D-F1B15FAA8EB2> /usr/lib/system/libsystem_trace.dylib
    0x7fff6f73f000 -     0x7fff6f744ffb  libunwind.dylib (35.4) <24A97A67-F017-3CFC-B0D0-6BD0224B1336> /usr/lib/system/libunwind.dylib
    0x7fff6f745000 -     0x7fff6f774fff  libxpc.dylib (1336.261.2) <7DEE2300-6D8E-3C00-9C63-E3E80D56B0C4> /usr/lib/system/libxpc.dylib
Sample analysis of process 35455 written to file /dev/stdout

```

kisvegabor, 2020-05-21:
I made a mistake when releasing the `lv_drivers` repo. Please update it from `master`.

zhepoch, 2020-05-21:
I'm checkout `lv_drivers` repo to master, And can show simulator window. Thinks.

closed 2020-05-21 by zhepoch

kisvegabor, 2020-05-22:
I've juggled a little with the `lv_drivers` could you try if it still works well?

zhepoch, 2020-05-25:
I'm tried, this works well.

kisvegabor, 2020-05-25:
Great! Thanks!