lvgl/lv_port_pc_eclipse #59: simulator crashes when LV_LAYOUT_ROW_MID is selected.

nusret3, 2020-05-24:
Line 123 in lv_examples/src/lv_demo_widgets/lv_demo_widgets.c
```
lv_cont_set_layout(h, LV_LAYOUT_PRETTY_MID); // works fine
// lv_cont_set_layout(h, LV_LAYOUT_ROW_MID); // TFT simulator crashes without any message
```

kisvegabor, 2020-05-25:
It's a tricky situation and unfortunately I don't know how to detect it.

The issue happens with the panel with text areas. The text areas have horizontal auto fit  `LV_FIT_PARENT`. So their sides are automatically set to the sides of the parent. With `LV_LAYOUT_ROW_MID` the following happens:
1. Create a text area (`ta1`)
2. LVGL sets its sides to the parent's left and right side
3. The row layout aligns it to the left.
4. Create another text area (`ta2`)
5. LVGL sets its sides to the parent's left and right side
6. The row layout aligns it next to `ta1`
7. LVGL sees that `ta2`'s sides are not at the parent's sides (because the layout moved it) so it sets `ta2`'s sides again. The layout is applied again. The sides are modified again... Infinite loop...

embeddedt, 2020-05-25:
If I understand this correctly, you can't satisfy both of these layouts, from a logical perspective. I think `LV_LAYOUT_ROW_MID` should take higher precedence in this case and `LV_FIT_PARENT` should be ignored.

An alternative option is to make `LV_FIT_PARENT` cause that container to get its own row.

nusret3, 2020-05-26:
PRETTY layout aligns the objects to the left and right sides of the parent. If the widgets are small it does not look pretty at all. So I started digging the header files to find out that how does other container layout options work. I don't think it goes to an infinite loop. I tried it on hardware as well and it traps at HardFault_Handler. ARM cores does not trap at HardFault in infinite loops. Probably it makes a division by zero or there might be a problem in pointer arithmetics. Besides this little problem the graphics stack works like a charm. I will keep this open a few more days if I find a workaround for this "known issue"

embeddedt, 2020-05-26:
It can enter a hard fault if there is a stack overflow, which always happens with infinite recursion.

nusret3, 2020-05-26:
Exactly @embeddedt. I agree with you. It does not seem to me an infinite loop. Stack overflow can trigger hardfault exception as well. @kisvegabor described it as infinite loop. Honestly since he is the author I did not debug it for a nested call that could cause a stack overflow.

kisvegabor, 2020-05-27:
Sorry for being unclear. To be more precise it's rather an infinite recursive loop.

The problem is that "fit" (applied to children) and "layout" (applied to parents) are handled independently. 

When the layout is applied it doesn't care with any property of the children just aligns them. "fit" is specific to `lv_cont` so it can't get easily.

Similarly, "fit" doesn't know about the layout of the parent because it's also specific to `lv_cont` and the children don't know the type of the parent.

embeddedt, 2020-05-27:
I think we may need to add some special casing where child containers can retrieve the parent's layout type (if it is a container) and parent containers can retrieve the child's fit type (if the child is a container).

kisvegabor, 2020-05-28:
I was thinking about making layout and fit to general `lv_obj` properties or style properties. 
This way e.g. a label's width could be set the parent's width.

I'd make `lv_cont` useless.

What do you think?

embeddedt, 2020-05-28:
I like that idea! HTML/CSS have something similar where there is no "special" container object.

We could keep the `lv_cont` API names around for backwards compatibility and then drop them entirely in 8.0.