lvgl/lv_port_pc_eclipse #61: [v7.0] Images rotated by default

BFFonseca, 2020-06-05:
Hi,
When I try to use images, they all show up slightly rotated by default.
Like this 
[image]

This is how I'm declaring them:
```lv_obj_t * img_var = lv_img_create(parent, NULL);
    lv_img_set_src(img_var, &icon_user);
    lv_obj_set_pos(img_var, 22, 5);

    img_var = lv_img_create(parent, NULL);
    lv_img_set_src(img_var, &icon_manual);
    lv_obj_set_pos(img_var, 256, 3);

    img_var = lv_img_create(parent, NULL);
    lv_img_set_src(img_var, &icon_signal);
    lv_obj_set_pos(img_var, 294, 7);
```
I've checked the 7.0 API and examples but nothing seems different.

This did not happen before migrating to 7.0, it worked fine. And I've re-uploaded the images just to make sure.

kisvegabor, 2020-06-05:
Hi,

Please attach `icon_user` too to reproduce the issue.

BFFonseca, 2020-06-05:
icon_user.c came directly from the LVGL online converter.
.c is not supported, renamed it to .txt. Here it is:
[icon_user.txt](https://github.com/lvgl/lv_sim_eclipse_sdl/files/4735642/icon_user.txt)

kisvegabor, 2020-06-05:
It works well for me. Can you try with only these lines:
```
  LV_IMG_DECLARE(icon_user);
  lv_obj_t * img_var = lv_img_create(lv_scr_act(), NULL);
  lv_img_set_src(img_var, &icon_user);
  lv_obj_set_pos(img_var, 22, 5);
```

BFFonseca, 2020-06-05:
I get the same results.

Initially I declared LV_IMG_DECLARE(icon_user) outside of that scope (in Static Variables), and use "parent" instead of lv_scr_act(). The image shows rotated.

Then I did as you sugested and passed LV_IMG_DECLARE(icon_user) into the same scope, replacing "parent" for lv_scr_act(). The image still shows rotated.

I then replaced lv_scr_act() with "parent" again and the image disappears.

I then "inverted" the conditions. declared LV_IMG_DECLARE(icon_user) on the global scope (as I had before - Static Variables) and used lv_scr_act() again. The image doesn't appear as well.

I suspect this has to do with something I did before messing with the img rotation. Can you think of anything that might interfere with the image rotation?

BFFonseca, 2020-06-05:
After some experiment, if I tilt the images at exactly a 334,4º angle they show correctly, although somewhat deformed due to the rotation.
They are tilted at a 25,6º angle. Not sure if this is any help.

kisvegabor, 2020-06-08:
It's very interesting because 25.6° is 256 with 0.1° resolution and `LV_IMG_ZOOM_NONE` (default value of zoom) is 256. Zoom and angle are always calculated together so maybe there is a typo somewhere. 

However, I also should be able to reproduce the issue with such a simple code. 

- Which version of LVGL do you use? (see lvgl.h)
- Which theme do you use?

BFFonseca, 2020-06-08:
I'm using v7.0.0 rc. The one on _dev-7.0_ branch on this repo.

While trying to find the issue, I did come across that function in lv_img.c, line 232 (lv_img_set_src):

```
    /*Provide enough room for the rotated corners*/
    if(ext->angle || ext->zoom != LV_IMG_ZOOM_NONE) lv_obj_refresh_ext_draw_pad(img);

    lv_obj_invalidate(img);
```

As for theme, I'm not sure. I haven't used them yet, I've only changed styles.

embeddedt, 2020-06-08:
Can you try it with the `master` branch? The `dev-7.0` branch no longer exists upstream. It probably won't change anything but it's worth a try.

BFFonseca, 2020-06-08:
Master branch is working no problem!
No weird rotations.

And it also fixed another problem I was having with pin mode on the text area (showing " instead of bulletpoints)

I'll stick with the master branch from now on then. Thought it was still in 6.2.

Thank you!

closed 2020-06-08 by BFFonseca

kisvegabor, 2020-06-08:
Great! Thank you for the feedback!

BFFonseca, 2020-06-08:
Thank you for your help @kisvegabor and @embeddedt !
I will be using LVGL for at least the next month.
I suspect I will find some more bugs or issues along the way.
Should I keep posting them here or should I prefer the Forum over GitHub?

embeddedt, 2020-06-08:
It's fine to post bug reports here on GitHub. Questions should be directed to the forum.