lvgl/lv_port_pc_eclipse #62: [v7.0] Keyboard - Can't input more than "max_length" total characters in password mode

BFFonseca, 2020-06-08:
I am trying to create a simple T9 keyboard to input a pin number (4 numbers). It has all 10 numbers and a backspace button (LV_SYMBOL_BACKSPACE).

I created a text area on this keyboard with the password mode enabled and a max length of 4 characters.

I have the following issue:
- The text area  only accepts a total of 4 characters, regardless if I delete any. 
So, if I input 4 numbers and then delete any, I can't write any more numbers on it. Unless I delete all the numbers, then I can input only the first number (this is constant - I can always input the number on the first position).

- If I input 3 numbers and delete one, I can input 1 more number (totaling 4 inputs) and then I cannot input any more numbers unless I delete every input. In which case I can only input on the first position, as before.

- If I input 2 numbers, delete one, input 1 again (3 total), delete one, input 1 (4 total) I reach the same state. Can't input any unless I delete all, then can only input on the first position.

**If I don't enable password mode**, everything works fine and I can always input up to 4 numbers, regardless if I delete any.
There seems to be an issue between **lv_textarea_set_max_length** and **lv_textarea_set_pwd_mode**.

Here's the code I'm running:
```
const char * map_pin_kb[] = {"1", "2", "3", "\n",
                                               "4", "5", "6", "\n",
                                               "7", "8", "9", "\n",
                                               " ", "0", LV_SYMBOL_BACKSPACE, ""
};
```

```
void create_pin_menu(lv_obj_t * parent)
        static lv_style_t style_pin_kb;
	lv_style_init(&style_pin_kb);
	lv_style_set_bg_color(&style_pin_kb, LV_STATE_DEFAULT, LV_COLOR_WHITE);
	lv_style_set_border_width(&style_pin_kb, LV_STATE_DEFAULT, 0);
	lv_style_set_text_font(&style_pin_kb, LV_STATE_DEFAULT, &lv_font_montserrat_26);

	lv_btnmatrix_ctrl_t kb_pin_ctrl_map[12] = {
            1, 1, 1,
            1, 1, 1,
            1, 1, 1,
            1 | LV_BTNMATRIX_CTRL_HIDDEN, 1, 1
	};

	lv_obj_t * kb_pin = lv_keyboard_create(parent, NULL);
	lv_keyboard_set_map(kb_pin, LV_KEYBOARD_MODE_NUM, map_pin_kb);
	lv_keyboard_set_mode(kb_pin, LV_KEYBOARD_MODE_NUM);
	lv_btnmatrix_set_ctrl_map(kb_pin, kb_pin_ctrl_map);
	lv_btnmatrix_set_one_check(kb_pin, true);
	lv_obj_add_style(kb_pin, LV_KEYBOARD_PART_BG, &style_pin_kb);
	lv_obj_add_style(kb_pin, LV_KEYBOARD_PART_BTN, &style_pin_kb);
	lv_obj_set_size(kb_pin, 180, 190);
	lv_obj_align(kb_pin, NULL, LV_ALIGN_CENTER, -50, 0);
	lv_keyboard_set_cursor_manage(kb_pin, true);

	static lv_style_t style_ta_pin;
	lv_style_init(&style_ta_pin);
	lv_style_set_border_width(&style_ta_pin, LV_STATE_DEFAULT, 3);
	lv_style_set_text_color(&style_ta_pin, LV_STATE_DEFAULT, LV_COLOR_BLUE);
	lv_style_set_text_font(&style_ta_pin, LV_STATE_DEFAULT, &lv_font_montserrat_24);

	lv_obj_t * ta_pin = lv_textarea_create(parent, NULL);
	lv_obj_add_style(ta_pin, LV_PAGE_PART_BG, &style_ta_pin);
	lv_obj_set_width(ta_pin, 80);
	lv_obj_align(ta_pin, kb_pin, LV_ALIGN_OUT_RIGHT_MID, 20, 4);
	lv_textarea_set_pwd_mode(ta_pin, true);
        lv_textarea_set_pwd_show_time(ta_pin, 0);
        lv_textarea_set_one_line(ta_pin, true);
	lv_textarea_set_text(ta_pin, "");
        lv_textarea_set_max_length(ta_pin, 4);

	lv_keyboard_set_textarea(kb_pin, ta_pin);
}
```

Notes: Running on the `master` branch of `lv_sim_eclipse_sdl` in Eclipse IDE, ubuntu 18.04

embeddedt, 2020-06-08:
There was a patch applied a few hours ago that may be related: https://github.com/lvgl/lvgl/commit/13a53d077de74dd87df609e448a7d7d307b42438

Have you updated your `lvgl` submodule to the tip of the LVGL `master` branch?

BFFonseca, 2020-06-08:
I'm not sure how to do that, I'm working based on master of this repo (the simulator one). 
Started working on git and this library about a couple weeks ago, still very green.

Still, I manually made the changes in that commit (`txt_label` to `ext->pwd_tmp`) and it is working correctly now. I got this issue about 4 hours ago, what timing!

Once again, thank you so much!

closed 2020-06-08 by BFFonseca

embeddedt, 2020-06-08:
> I'm not sure how to do that, I'm working based on master of this repo (the simulator one).
Started working on git and this library about a couple weeks ago, still very green.

If you open the `lvgl` directory in a terminal, run `git pull`, `git checkout master`, and `git pull` a second time and it should update.