lvgl/lv_port_pc_eclipse #63: change default keyboard

mostafajalal20, 2020-07-12:
hi, how can i change default keyboard from English to another language such as persian, because i read the code and realize that the code have persian keyboard and font ,please help me

embeddedt, 2020-07-12:
I don't see anything in the core library related to a Persian keyboard; where did you see this?

mostafajalal20, 2020-07-12:
in lvgl/src/lv_misc/lv_txt_ap.c

embeddedt, 2020-07-12:
Oh, that's just for handling the rendering of Persian characters. There isn't a built-in Persian keyboard layout; you'd have to create one yourself.

mostafajalal20, 2020-07-12:
do you have any document or guidance to help me for this?

kisvegabor, 2020-07-13:
See here: https://docs.lvgl.io/latest/en/html/widgets/keyboard.html#new-keymap

mostafajalal20, 2020-07-13:
thanks a lot

closed 2020-07-14 by kisvegabor