lvgl/lv_port_pc_eclipse #66: Cross compiling demo application failed

shree-97, 2020-08-03:
Hi,
I have downloaded the PC simulator and I was able to run the demo application on PC. But my end goal was to run the demo application on i.MX6. So, for that I was trying to cross compile the PC simulator project but it was failed. The error is as below:
l**v_drivers/display/monitor.c:19:34: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
Makefile:45: recipe for target 'lv_drivers/display/monitor.o' failed
make: *** [lv_drivers/display/monitor.o] Error 1**

embeddedt, 2020-08-03:
You will need an iMX6 version of SDL installed and available to your cross compiler. Personally I'd suggest using the [Linux framebuffer](https://github.com/lvgl/lv_port_linux_frame_buffer) project for iMX6 as it doesn't need SDL installed.

shree-97, 2020-08-04:
Hi,
Thanks for your suggestion. I did try the linux framebuffer project and I was able to see the demo app running on screen but touchscreen is not working even after following the instructions.

I do have some questions which are as below:
1) If my display is interfaced with 18-bit then what color depth should I use ?
2) Is there any possibility of integrating Amazon Alexa with LVGL ?

embeddedt, 2020-08-04:
> If my display is interfaced with 18-bit then what color depth should I use ?

Either 16bpp or 32bpp should be fine. I believe the Linux driver is smart enough to translate to the screen's actual color depth.

> Is there any possibility of integrating Amazon Alexa with LVGL ?

I'm not quite sure what you mean by this. LVGL is a GUI framework - you can integrate anything you wish with it.

shree-97, 2020-08-04:
I mean is it possible to navigate the GUI through voice commands instead of touchscreen or buttons ?

embeddedt, 2020-08-04:
LVGL gives you a framework for building user interfaces. Voice input is not part of the core library, but could plausibly be implemented by focusing/switching between GUI screens as needed.

closed 2020-09-05 by stale[bot]