lvgl/lv_port_pc_eclipse #67: lvgl/lvgl.h : no such file or directory

PCov3r, 2020-08-06:
Hello everyone,
I tried installing the pc simulator for windows. I firstly tried to make it work on eclipse and got the following error : 
`23:25:14 **** Incremental Build of configuration Debug for project lv_sim_eclipse_sdl-master ****
make all 
'Building file: ../main.c'
'Invoking: Cross GCC Compiler'
gcc -I"C:\Users\pauld\Documents\Eclipse Workspace\lv_sim_eclipse_sdl-master" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -MMD -MP -MF"main.d" -MT"main.o" -o "main.o" "../main.c"
../main.c:16:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [subdir.mk:23: main.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

23:25:19 Build Failed. 2 errors, 0 warnings. (took 4s.473ms)
`

I then tried to make it work on codeblocks (as I read on an other issue that it was easier to make it work). But same problem again as you can read : 

`||=== Build: Debug in LittlevGL (compiler: GNU GCC Compiler) ===|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\display\fbdev.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\display\monitor.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\display\R61581.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\display\SSD1963.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\display\ST7565.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\AD_touch.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\evdev.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\FT5406EE8.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\keyboard.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\mouse.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\indev\XPT2046.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_drivers\win_drv.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\benchmark\benchmark.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\benchmark\benchmark_bg.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\demo\demo.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\demo\img_bubble_pattern.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\sysmon\sysmon.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\terminal\terminal.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_apps\tpcal\tpcal.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_group\lv_test_group.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_misc\lv_test_task.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_obj\lv_test_obj.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_arc\lv_test_arc.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_bar\lv_test_bar.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_btn\lv_test_btn.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_btnm\lv_test_btnm.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_canvas\lv_test_canvas.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_cb\lv_test_cb.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_chart\lv_test_chart.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_cont\lv_test_cont.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_ddlist\lv_test_ddlist.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_gauge\lv_test_gauge.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_img\img_flower_icon.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_img\lv_test_img.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_imgbtn\imgbtn_img_1.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_imgbtn\imgbtn_img_2.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_imgbtn\imgbtn_img_3.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_imgbtn\imgbtn_img_4.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_imgbtn\lv_test_imgbtn.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_kb\lv_test_kb.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_label\lv_test_label.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_led\lv_test_led.c|
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lv_examples\lv_tests\lv_test_objx\lv_test_line\lv_test_line.c|
[... cut ...]
||WARNING: Can't read file's timestamp: C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\lvgl\src\lv_themes\lv_theme_zen.c|
C:\Users\pauld\Downloads\lv_sim_codeblocks_win-master\lv_sim_codeblocks_win-master\main.c|13|fatal error: lvgl/lvgl.h: No such file or directory|
||=== Build failed: 1 error(s), 166 warning(s) (0 minute(s), 0 second(s)) ===|
`

I must say I'm a bit of a newbie here, I mostly code on the arduino ide or in python. What did I do wrong ?

Thanks !

embeddedt, 2020-08-06:
You need to clone the CodeBlocks repository and update submodules. Downloading a ZIP from GitHub won't work.

`git clone --recursive https://github.com/lvgl/lv_sim_codeblocks_win.git`

PCov3r, 2020-08-07:
Ho ! Thanks. It works now !

closed 2020-08-07 by PCov3r