lvgl/lv_port_pc_eclipse #68: Error when trying to run on Eclipse (macOS)

david-res, 2020-08-09:
It compiles fine, but when I run it I get the following error
[image]
What am I missing?

Im using a 2016 Macbook Pro with the latest OS on Eclipse CDT

embeddedt, 2020-08-09:
I believe the project name changed but the Eclipse launch configuration wasn't updated. You can try editing the launch configuration and choosing the correct binary (`Debug/lv_sim_eclipse_sdl`, if I recall correctly).

david-res, 2020-08-09:
@embeddedt that fixed it, thank you!

embeddedt, 2020-08-09:
Glad to hear it. With https://github.com/lvgl/lv_sim_eclipse_sdl/commit/28f8ed4e60e44cba63cfb2302550a0b7074e8cff it should be fixed upstream as well.

commit 28f8ed4e (2020-08-09): Fix launch configuration (closes #68)
closed 2020-08-09 by embeddedt