lvgl/lv_port_pc_eclipse #69: Multiple definitions of main when updating to LVGL 7.3.1

BFFonseca, 2020-08-20:
When I use LVGL 7.3.1 in the simulator I get a build error saying there are multiple definitions of main.
From what I can tell this is due to the "file(GLOB_RECURSE ...)" in the CMakeLists.txt file.

I can't figure out a way to fix this.

```
./build/CMakeFiles/feature_tests.o: In function `main':
/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../build/CMakeFiles/feature_tests.c:34: multiple definition of `main'
./build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.o:/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c:574: first defined here
./main.o: In function `main':
makefile:156: recipe for target 'lv_sim_eclipse_sdl' failed
/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../main.c:51: multiple definition of `main'
./build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.o:/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c:574: first defined here
collect2: error: ld returned 1 exit status
make: *** [lv_sim_eclipse_sdl] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
```

In my CMakeLists.txt file I have:

>  cmake_minimum_required(VERSION 3.10)
> project(lvgl)
> set(CMAKE_C_STANDARD 11)#C11
> set(CMAKE_CXX_STANDARD 17)#C17
> set(CMAKE_CXX_STANDARD_REQUIRED ON)
> 
> INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR})
> 
> file(GLOB_RECURSE INCLUDES "lv_drivers/*.h" "lv_examples/*.h"  "lvgl/*.h"  "./*.h" "smart_ring_ui/*.h" "smart_ring_ui/menus/*.h")
> file(GLOB_RECURSE SOURCES  "lv_drivers/*.c" "lv_examples/*.c"  "lvgl/*.c"  "smart_ring_ui/*.c" "smart_ring_ui/menus/*.c")
> file(REMOVE_ITEM SOURCES "Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c")
> file(REMOVE_ITEM SOURCES "/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c")
> list(REMOVE_ITEM INCLUDES ${REMOVE_GLOB})
> 
> SET(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin) 
> 
> find_package(SDL2 REQUIRED SDL2)
> include_directories(${SDL2_INCLUDE_DIRS})
> add_executable(main main.c mouse_cursor_icon.c ${SOURCES} ${INCLUDES})
> target_link_libraries(main PRIVATE SDL2 )
> add_custom_target (run COMMAND ${EXECUTABLE_OUTPUT_PATH}/main)
> 

I added these lines in hope to fix this, but it didn't.

> file(REMOVE_ITEM SOURCES "Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c")
> file(REMOVE_ITEM SOURCES "/home/bfonseca/GitHub/lv_sim_eclipse_sdl_master/Debug/../build/CMakeFiles/3.10.2/CompilerIdC/CMakeCCompilerId.c")

My experience with CMake is very small so I am not sure what else I should do/try.

From what I've read listing files manually is the best practice instead of using GLOB. Guessing this isn't exactly viable.

closed 2020-08-20 by BFFonseca

kisvegabor, 2020-08-24:
No problem.

I think issues can't be deleted. :slightly_frowning_face:

BFFonseca, 2020-08-24:
> I think issues can't be deleted.

Only repository Admins can delete issues. On the right hand side, bottom option.

kisvegabor, 2020-08-24:
I'm admin, but the issue can be only locked:
[image]

BFFonseca, 2020-08-24:
> I'm admin, but the issue can be only locked:

That's weird... I can delete them in my private repositories :thinking:

kisvegabor, 2020-08-24:
Ah, it was required to be enabled in the organization.