lvgl/lv_port_pc_eclipse #73: #define LV_USE_USER_DATA twice in lv_conf.h

zhaoyk, 2020-10-15:
https://github.com/lvgl/lv_sim_eclipse_sdl/blob/master/lv_conf.h#L532

I guess it should be `#define LV_USE_USER_DATA_FREE 1` ?

embeddedt, 2020-10-15:
Correct.

@kisvegabor Any idea why the script did this?

kisvegabor, 2020-10-16:
It was really an issue in the release script. When searched for defines in `lv_conf_templ.h` to create the project-specific `lv_conf.h` it didn't check the end of the define name end. So both `LV_USE_USER_DATA` and  `LV_USE_USER_DATA_FREE` matched  `LV_USE_USER_DATA`. Therefore both ware replaced.

It is fixed here
https://github.com/lvgl/lvgl/commit/8d9c0bd9b2e526b7d8bce35b868c2d37612fb3f0

kisvegabor, 2020-11-08:
Looks good in the new release so I close this issue:
https://github.com/lvgl/lv_sim_eclipse_sdl/blob/master/lv_conf.h#L546

Thank for the report!

closed 2020-11-08 by kisvegabor