lvgl/lv_port_pc_eclipse #76: lv_sim_vscode_sdl

microwavesafe, 2020-11-13:
I've created a VSCode project for running LVGL on SDL. I don't use Eclipse, so it was useful to me.

I hand edited the Makefile as that is what I'm familiar with, rather than go the CMake route. Not sure if that makes a difference. Anyway you can just `git clone --recursive https://github.com/microwavesafe/lv_sim_vscode_sdl` the repository, open the workspace and hit F5 to start debugging the example UI.

I'm more than happy for you to take control of the project if it's useful to you.

kisvegabor, 2020-11-16:
Hi,

That's awesome! I've already dreamed a lot about a VSCode simulator project. :slightly_smiling_face: 

So I'd be happy to host your project here. Of course, you would have admin rights to the repo.

microwavesafe, 2020-11-16:
I think I can transfer ownership to you through the settings, I've not done this before, is there anything I need to do before that?

kisvegabor, 2020-11-16:
I also did it only with organizations where I was admin in both, but there wasn't anyhing special.  
I'll simply get a notification about the transfer query.

microwavesafe, 2020-11-16:
I get this message when I try

```You don’t have the permission to create public repositories on lvgl ```

kisvegabor, 2020-11-16:
It says you need to have "repository creation permissions" in the target organization. 
https://help.github.jp/enterprise/2.11/user/articles/transferring-a-repository-owned-by-your-personal-account/

kisvegabor, 2020-11-16:
I've sent an invitation to temporarily give you "Member access" to the organization.

microwavesafe, 2020-11-16:
OK that worked, it says it is transferring and may take a few minutes.

kisvegabor, 2020-11-16:
Great, it already here: https://github.com/lvgl/lv_sim_vscode_sdl
Is it finished on your side too?

kisvegabor, 2020-11-16:
I see that the commit came from [springltd](https://github.com/springltd). Which user shall give access in the new repo?

kisvegabor, 2020-11-16:
I've just got a confirmation email about the transfer was successful so withdrawn the access to the whole organization.

microwavesafe, 2020-11-16:
Yep it's all done here. Sorry that's my work account. I never found a simple way of swapping accounts so I tend to just leave everything as my work account. Add my personal account and I will fix how I access it.

kisvegabor, 2020-11-16:
I've invited @microwavesafe as admin. 
I can also add @springltd if you wish. From our side, it doesn't really matter if both accounts are yours.

microwavesafe, 2020-11-16:
Actually that would be useful if you don't mind. I've updated the README with the new url.

kisvegabor, 2020-11-16:
> Actually that would be useful if you don't mind.

Added. 

>  I've updated the README with the new url.

Thanks!

closed 2020-11-16 by microwavesafe