lvgl/lv_port_pc_eclipse #81: Docker build is not working on changeset on master head @e151ce7

lyorak, 2020-12-20:
Git submodule of lv_examples is referring to a changeset that has a warning in lv_examples/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.c and hence won't build. I see that the issue has been fixed on a later version of ll_examples, should it be bumped in git submodules?

lyorak, 2020-12-20:
See failing docker build here https://github.com/lyorak/lv_sim_eclipse_sdl/runs/1585216662?check_suite_focus=true

kisvegabor, 2020-12-21:
@babldev could you take look at it?

embeddedt, 2020-12-21:
> And this repo was changed from the lv_examples master to 7.8.1.

That was probably caused by the release script.

> Let me know what the best path forward here.

It seems to me that it would be simplest to just release 7.8.2.

babldev, 2020-12-21:
I agree, sounds like a plan

babldev, 2020-12-21:
A note going forward on this approach is we shouldn't directly modify the submodules, and instead cut a relesae

kisvegabor, 2020-12-22:
Thanks for checking it.

I've just updated `lv_examples` to `master`. It will be part of the next release.

lyorak, 2020-12-23:
As I see updating the submodule itself did not solve the issue, just ended up in another compilation error. A 
See https://github.com/lyorak/lv_sim_eclipse_sdl/runs/1600083761?check_suite_focus=true 
The macro LV_USE_DEMO_MUSIC is missing from lv_ex_conf.h

kisvegabor, 2020-12-27:
@lyorak Added: https://github.com/lvgl/lv_sim_eclipse_sdl/commit/0521232b720e8c38069668e36318c793e4f3dc97

embeddedt, 2021-01-19:
@kisvegabor Something is wrong with the update script. The commit message says "Update to 7.9.0" but the actual [`release/v7`]( https://github.com/lvgl/lv_examples/tree/release/v7) branch on `lv_examples` is stuck on a commit from the beginning of December. I think that is part of the problem here.

kisvegabor, 2021-01-19:
I've added a fix after the last release. It failed to correctly update the repo if there were no changes. 
We will see how it works with today's release.

kisvegabor, 2021-04-13:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2021-04-13 by kisvegabor