lvgl/lv_port_pc_eclipse #82: Consider deprecating the dedicated Makefile?

embeddedt, 2020-12-26:
There are currently 3 ways to build this project (not counting Docker):

* Eclipse IDE.
* The dedicated Makefile in the root directory.
* CMake.

CMake has proliferated throughout the open-source community as of late and I see it being used in many different projects. As such, it is a component which a developer is likely to have on their system. It is also more reliable at detecting changes to dependencies than our current Makefile and is faster at rebuilding the project.

Given the above, I see no reason to keep the dedicated Makefile, and would like to propose removing it with the v8 release.

kisvegabor, 2020-12-29:
I have no objection if we provide a step by step tutorial in the README to describe how to build the project with CMake.

embeddedt, 2020-12-29:
How does this sound? https://github.com/lvgl/lv_sim_eclipse_sdl/blob/master/README.md#cmake

kisvegabor, 2020-12-30:
Great!

The only thing is that `cmake --build . --parallel` was not working for me with `3.10.2` cmake and Linux Mint:
```
Unknown argument --parallel
```

Anyway, I think it's a local issue so let's remove the Makefile. 

BTW, I was impressed by how much faster cmake is than the Makefile.

embeddedt, 2020-12-30:
> not working for me with`3.10.2` cmake and Linux Mint:

The feature was added in CMake 3.12 according to [this SO answer](https://stackoverflow.com/a/50883540). 3.10 is still the default version on Ubuntu 18.04, so do you think we should remove `--parallel` for now?

> I was impressed by how much faster cmake is than the Makefile.

I thought the same thing! I guess Make is quite slow at processing dependencies.

kisvegabor, 2020-12-30:
> The feature was added in CMake 3.12 according to this SO answer. 3.10 is still the default version on Ubuntu 18.04, so do you think we should remove --parallel for now?

Ubuntu 20.04 is already out so I think we can keep it. But maybe a note can be added like "--parallel is supported from cmake v3.12"

embeddedt, 2021-01-21:
Alright; I added a note in the README and the Makefile. I'll remove the latter when v8 is released.

closed 2021-01-21 by embeddedt

kisvegabor, 2021-01-21:
Great!