lvgl/lv_port_pc_eclipse #83: error declaration of ‘ta’ shadows a global declaration [-Werror=shadow]

mourazo, 2021-01-17:
I'm trying to build it, but the following error appears:

> make all 
> lv_examples/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.c: In function ‘ta_event_cb’:
> lv_examples/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.c:27:36: error: declaration of ‘ta’ shadows a global declaration [-Werror=shadow]
>    27 | static void ta_event_cb(lv_obj_t * ta, lv_event_t e)
>       |                         ~~~~~~~~~~~^~
> lv_examples/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.c:5:19: note: shadowed declaration is here
>     5 | static lv_obj_t * ta;
>       |                   ^~
> cc1: all warnings being treated as errors
> make: *** [Makefile:45: lv_examples/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.o] Error 1
> "make all" terminated with exit code 2. Build might be incomplete.

embeddedt, 2021-01-17:
Have you updated to the latest version of the project? https://github.com/lvgl/lv_examples/pull/86 should have already fixed this problem, and should be included in the latest version.

mourazo, 2021-01-17:
Yes, it's not included in the latest version of the project if you clone with 

> git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git

I updated into this project the lv_examples but now I get the next error: 

> make all 
> lv_examples/src/lv_ex_widgets/lv_ex_img/lv_ex_img_3.c: In function 'lv_ex_img_3':
> lv_examples/src/lv_ex_widgets/lv_ex_img/lv_ex_img_3.c:23:29: error: cast between incompatible function types from 'void (*)(lv_obj_t *, uint16_t)' {aka 'void (*)(struct _lv_obj_t *, short unsigned int)'} to 'void (*)(void *, lv_anim_value_t)' {aka 'void (*)(void *, short int)'} [-Werror=cast-function-type]
>    23 |     lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_img_set_zoom);
>       |                             ^
> cc1: all warnings being treated as errors
> make: *** [Makefile:45: lv_examples/src/lv_ex_widgets/lv_ex_img/lv_ex_img_3.o] Error 1
> "make all" terminated with exit code 2. Build might be incomplete.

embeddedt, 2021-01-19:
I can confirm there are issues with the submodules; let's move the conversation to #81 as the issue is similar.

closed 2021-01-19 by embeddedt