lvgl/lv_port_pc_eclipse #87: Windows ecplise build errors

Noneghost, 2021-02-04:
./lv_drivers/display/UC1610.o ./lv_drivers/display/drm.o ./lv_drivers/display/fbdev.o ./lv_drivers/display/monitor.o  ./lv_drivers/win_drv.o  ./main.o ./mouse_cursor_icon.o   -lmingw32 -lSDL2main -lSD

H:\LVGL\lv_sim_eclipse_sdl-master\Debug/../lv_drivers/display/monitor.c:238: undefined reference to `SDL_PollEvent'
./lv_drivers/display/monitor.o: In function `monitor_sdl_clean_up':
H:\LVGL\lv_sim

Hi, i use window`s ecplise projects , build don't success . I've linked the appropriate library, but there was an undefined error when linking.

kisvegabor, 2021-02-04:
`-lSD` should be `-lSDL2`.
Have you only missed the last two letters while copy/pasting?

Noneghost, 2021-02-05:
sorry,this copy miss .  i use link -lmingw32 -lSDL2main -lSD2

kisvegabor, 2021-02-08:
I had similar issues when used 32 SDL with 64 bit program or vice versa.

BTW, do you get only this error message or a lot of other undefined references to SDL functions?

kisvegabor, 2021-04-13:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2021-04-13 by kisvegabor