lvgl/lv_port_pc_eclipse #88: Error cloning the submodule gui

alexkalmuk, 2021-02-12:
I perform git clone as described in [README](https://github.com/lvgl/lv_sim_eclipse_sdl#get-the-pc-project):

```
git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git
```

ant getting the following error:

```
fatal: clone of 'https://github.com/lvgl-private/malmoset-hvac.git' into submodule path '/home/alexander/git/lv_sim_eclipse_sdl/gui' failed
```

I see the project is under `lvgl-private`, so maybe it's the reason, but I am not sure.

embeddedt, 2021-02-12:
That's a bug; I don't think the `gui` submodule is supposed to be in this repository. For now I've removed it.

alexkalmuk, 2021-02-12:
@embeddedt Thank you. I also removed `gui` module locally, but then got the compilating error:

```
/usr/bin/ld: CMakeFiles/main.dir/main.c.o: in function `main':
main.c:(.text+0x36): undefined reference to `gui_init'
collect2: error: ld returned 1 exit status
```

Is it safe to remove this `gui_init` from main.c?

embeddedt, 2021-02-12:
Yes.

alexkalmuk, 2021-02-12:
Thanks, now it's built successfully.

closed 2021-02-12 by alexkalmuk

alexkalmuk, 2021-02-12:
Ah, it's because of the recent [commit](https://github.com/lvgl/lv_sim_eclipse_sdl/commit/e90c9fab9d4d74b23dd3ff75d09d4f8d47a8a7a0). After I uncommented `lv_demo_widgets` in main.c things look pretty. But probably it's my mistake and I had to try the release version first :)

kisvegabor, 2021-02-12:
Sorry, my bad. I accidentally added the gui repo that pointed to private repo.

@embeddedt Thanks for the removal.

alexkalmuk, 2021-02-12:
No problem, thanks for your fixes.