lvgl/lv_port_pc_eclipse #90: small modification

uraich, 2021-03-16:
I got a compilation error in win32drh.h, saying that ../lv_drv_conf.h is missing.
In fact it is at position ../../lv_drv_conf.h
I corrected this in my version to make it compile. Then the demo runs fine.

Kealina-A, 2021-03-27:
Thanks. It work to me.

embeddedt, 2021-04-25:
This should be fixed now, since `LV_CONF_INCLUDE_SIMPLE` was added to both build systems. Let me know if it's still causing problems.

closed 2021-04-25 by embeddedt

uraich, 2021-04-28:
Sorry, couldn't test it. I pulled the newest changes of lvgl and this breaks several things.
LV_COLOR_PALETTE_LIGHT_BLUE is now called LV_PALETTE_LIGHT_BLUE 
lv_img_create now only has a single parameter
lv_theme_default_init now has a parameter more.
Trying to fix these on my version of the simulator

kisvegabor, 2021-04-28:
I've just updated the project.

uraich, 2021-04-28:
Still a question on this: There are now 2 sources for the examples, which are slightly different: lv_examples, which is used in lv_sim_eclipse and the examples in the lvgl repository. Are these going to be merged? and which set should I use for my translation to Python?
@embeddedt : You said you are going to provide V8 docs? Did you manage up upload and if yes, where can I find them. It is ok if I have to compile the docs

embeddedt, 2021-04-28:
The widget examples are now in the main `lvgl` repository. `lv_examples` now contains demos only.

v8 docs are currently being hosted at the link below, and will be moved to the normal location once I finish working on the script.

http://docs_compiled.lvgl.io/8.0/

uraich, 2021-04-28:
I am still fighting with lv_sim_eclipse_sdl
I clone  lv_sim_eclipse_sdl and check out the dev branch. Then I get the submodules. I make sure that lvgl is on the master branch.
I create a build directory, cd into it and run cmake ..
Then I run make.
This gives me errors like:
‘LV_HOR_RES_MAX’ undeclared here
error: unknown type name ‘lv_task_t’
error: ‘sdl_event_handler’ undeclared
etc.
What I am doing wrong?

embeddedt, 2021-04-28:
@uraich I have just tried your steps and it seemed to work fine for me. Perhaps you missed pulling the changes by accident?

uraich, 2021-04-29:
In fact I found out that lv_drivers was associated with no branch. git checkout master in lv_drivers did the trick. After this everything compiles fine.
As you can imagine it is extremely helpful when translating the C examples to Python that now I can see, how the original version is behaving.