lvgl/lv_port_pc_eclipse #91: Not building in Linux

nilacuna, 2021-03-23:
The README.MD instructions are clear enough to be followed. I am however getting this error below when compiling from within eclipse. 

Building file: ../lv_drivers/win32drv/win32drv.c
Invoking: Cross GCC Compiler
gcc -I"/mnt/ext4data/git/pc_simulator_sdl_eclipse" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -MMD -MP -MF"lv_drivers/win32drv/win32drv.d" -MT"lv_drivers/win32drv/win32drv.o" -o "lv_drivers/win32drv/win32drv.o" "../lv_drivers/win32drv/win32drv.c"
In file included from ../lv_drivers/win32drv/win32drv.c:10:
../lv_drivers/win32drv/win32drv.h:17:10: fatal error: ../lv_drv_conf.h: No such file or directory
   17 | #include "../lv_drv_conf.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [lv_drivers/win32drv/subdir.mk:20: lv_drivers/win32drv/win32drv.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

any comments?

nilacuna, 2021-03-23:
seems this is related to issue 90 (https://github.com/lvgl/lv_sim_eclipse_sdl/issues/90) I followed his suggestion and the project compiles okay now.

kisvegabor, 2021-03-23:
Does it work if you manually pull  `lv_drivers`?

nilacuna, 2021-03-23:
Hi. thank you for getting back to me. Still encountering the same problems. I mainly just removed --recursive and do manual update per submodule. The log for lv_driver below.

pci@pci-G3-3579:~/work/git/lv_sim_eclipse_sdl$ git submodule update --init lv_drivers
Submodule 'lv_drivers' (https://github.com/littlevgl/lv_drivers.git) registered for path 'lv_drivers'
Cloning into '/mnt/ext4data/git/lv_sim_eclipse_sdl/lv_drivers'...
Submodule path 'lv_drivers': checked out '8c96359f4199297809ab205870eb603b03d98b4e'

embeddedt, 2021-03-23:
Please try going into the `lv_drivers` directory and running `git checkout master` followed by `git pull`.

nilacuna, 2021-03-23:
@embeddedt I did exactly what you said (working from a fresh clone) and encountered an entirely different problem now. please see log of actions below:

pci@pci-G3-3579:~/work/git$ git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git
Cloning into 'pc_simulator_sdl_eclipse'...
remote: Enumerating objects: 132, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (105/105), done.
remote: Total 1343 (delta 71), reused 77 (delta 27), pack-reused 1211
Receiving objects: 100% (1343/1343), 3.01 MiB | 477.00 KiB/s, done.
Resolving deltas: 100% (727/727), done.
Submodule 'lv_drivers' (https://github.com/littlevgl/lv_drivers.git) registered for path 'lv_drivers'
Submodule 'lv_examples' (https://github.com/littlevgl/lv_examples.git) registered for path 'lv_examples'
Submodule 'lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'lvgl'
Cloning into '/mnt/ext4data/git/pc_simulator_sdl_eclipse/lv_drivers'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 1385 (delta 17), reused 22 (delta 6), pack-reused 1339
Receiving objects: 100% (1385/1385), 355.81 KiB | 216.00 KiB/s, done.
Resolving deltas: 100% (900/900), done.
Cloning into '/mnt/ext4data/git/pc_simulator_sdl_eclipse/lv_examples'...
remote: Enumerating objects: 1303, done.
remote: Counting objects: 100% (1303/1303), done.
remote: Compressing objects: 100% (569/569), done.
remote: Total 12862 (delta 895), reused 1015 (delta 720), pack-reused 11559
Receiving objects: 100% (12862/12862), 116.88 MiB | 519.00 KiB/s, done.
Resolving deltas: 100% (9043/9043), done.
Cloning into '/mnt/ext4data/git/pc_simulator_sdl_eclipse/lvgl'...
remote: Enumerating objects: 46091, done.
remote: Total 46091 (delta 0), reused 0 (delta 0), pack-reused 46091
Receiving objects: 100% (46091/46091), 39.81 MiB | 614.00 KiB/s, done.
Resolving deltas: 100% (36694/36694), done.
Submodule path 'lv_drivers': checked out '8c96359f4199297809ab205870eb603b03d98b4e'
Submodule path 'lv_examples': checked out '5f685373743f6999a5b6c0f2782767f2815e151b'
Submodule path 'lvgl': checked out 'e1ac174e166f07def7deccaf3b3e8e6b8f658a20'
pci@pci-G3-3579:~/work/git$ cd pc_simulator_sdl_eclipse/

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse$ cd lv_drivers/

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse/lv_drivers$ git branch
* (HEAD detached at 8c96359)
  master

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse/lv_drivers$ git checkout master
Previous HEAD position was 8c96359 Release v7.11.0
Switched to branch 'master'
Your branch is up to date with 'origin/master'.

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse/lv_drivers$ git pull
warning: Pulling without specifying how to reconcile divergent branches is
discouraged. You can squelch this message by running one of the following
commands sometime before your next pull:

  git config pull.rebase false  # merge (the default strategy)
  git config pull.rebase true   # rebase
  git config pull.ff only       # fast-forward only

You can replace "git config" with "git config --global" to set a default
preference for all repositories. You can also pass --rebase, --no-rebase,
or --ff-only on the command line to override the configured default per
invocation.

Already up to date.

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse/lv_drivers$ git branch
* master

------------------------- see new compile error below ---------------------

Building file: ../lv_drivers/indev/keyboard.c
Invoking: Cross GCC Compiler
gcc -I"/mnt/ext4data/git/pc_simulator_sdl_eclipse" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -MMD -MP -MF"lv_drivers/indev/keyboard.d" -MT"lv_drivers/indev/keyboard.o" -o "lv_drivers/indev/keyboard.o" "../lv_drivers/indev/keyboard.c"
../lv_drivers/indev/keyboard.c: In function ‘keyboard_handler’:
../lv_drivers/indev/keyboard.c:72:21: error: ‘LV_INDEV_STATE_PRESSED’ undeclared (first use in this function); did you mean ‘LV_BTN_STATE_PRESSED’?
   72 |             state = LV_INDEV_STATE_PRESSED;          /*Save the key is pressed now*/
      |                     ^~~~~~~~~~~~~~~~~~~~~~
      |                     LV_BTN_STATE_PRESSED
../lv_drivers/indev/keyboard.c:72:21: note: each undeclared identifier is reported only once for each function it appears in
../lv_drivers/indev/keyboard.c:75:21: error: ‘LV_INDEV_STATE_RELEASED’ undeclared (first use in this function); did you mean ‘LV_BTN_STATE_RELEASED’?
   75 |             state = LV_INDEV_STATE_RELEASED;         /*Save the key is released but keep the last key*/
      |                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                     LV_BTN_STATE_RELEASED
make: *** [lv_drivers/indev/subdir.mk:41: lv_drivers/indev/keyboard.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

07:24:02 Build Failed. 3 errors, 1 warnings. (took 18s.97ms)

embeddedt, 2021-03-23:
Ah... that's because this is still a v7 repo and v8 got merged into `master` in `lv_drivers`. Checkout the `release/v7` branch of `lv_drivers` instead.

nilacuna, 2021-03-23:
@embeddedt 

pci@pci-G3-3579:~/work/git/pc_simulator_sdl_eclipse/lv_drivers$ git checkout -b v7 origin/release/v7
Branch 'v7' set up to track remote branch 'release/v7' from 'origin'.
Switched to a new branch 'v7'

now I am back to my old problem:

Building file: ../lv_examples/src/assets/lv_font_montserrat_28_compr_az.c
Invoking: Cross GCC Compiler
gcc -I"/mnt/ext4data/git/pc_simulator_sdl_eclipse" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -MMD -MP -MF"lv_examples/src/assets/lv_font_montserrat_28_compr_az.d" -MT"lv_examples/src/assets/lv_font_montserrat_28_compr_az.o" -o "lv_examples/src/assets/lv_font_montserrat_28_compr_az.o" "../lv_examples/src/assets/lv_font_montserrat_28_compr_az.c"
Finished building: ../lv_examples/src/assets/lv_font_montserrat_28_compr_az.c
 
Building file: ../lv_drivers/win32drv/win32drv.c
Invoking: Cross GCC Compiler
gcc -I"/mnt/ext4data/git/pc_simulator_sdl_eclipse" -O0 -g3 -Wall -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -MMD -MP -MF"lv_drivers/win32drv/win32drv.d" -MT"lv_drivers/win32drv/win32drv.o" -o "lv_drivers/win32drv/win32drv.o" "../lv_drivers/win32drv/win32drv.c"
In file included from ../lv_drivers/win32drv/win32drv.c:10:
../lv_drivers/win32drv/win32drv.h:17:10: fatal error: ../lv_drv_conf.h: No such file or directory
   17 | #include "../lv_drv_conf.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [lv_drivers/win32drv/subdir.mk:20: lv_drivers/win32drv/win32drv.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

embeddedt, 2021-03-23:
I was testing using CMake, which apparently hid the problem. The latest commit to this repo should fix the problem and the extra `lv_drivers` manipulation should not be needed.

Let me know if it's still broken.

closed 2021-03-23 by embeddedt

nilacuna, 2021-03-24:
@embeddedt works out of the box now. thank you for your time fixing this.

jeremyherbert, 2021-03-28:
I just did a brand new checkout and am still having this issue

embeddedt, 2021-03-28:
This is a good reminder to add a CI to this repository. :wink: The problem should be fixed if you pull again.

jeremyherbert, 2021-03-28:
I can confirm it works now. Thanks for your help.

commit de406ef3 (2021-03-23): Use correct include style when building using Eclipse (should fix #91)