lvgl/lv_port_pc_eclipse #92: Scrollbars not working

jeremyherbert, 2021-03-28:
Hi,

I just did a checkout and build of the latest version. In the TFT Simulator window, I have scroll bars visible on the "Visuals" and "Selectors" page, but I can't actually use them to do anything. I have tried using the mouse to click and drag the scroll bar, using my mouse scroll wheel and using the up and down arrow keys, but none of them do anything. I can see there are some widgets offscreen in the selectors page, so I am assuming that the scrollbar is indeed supposed to work.

I'm running Xubuntu 20.04:
```Linux ubuntu 5.4.0-70-generic #78-Ubuntu SMP Fri Mar 19 13:29:52 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux```

Latest commit is: de406ef3cf83197528ce7f347e2eb8ddf6feb14d

jeremyherbert, 2021-03-28:
Sorry, looks like I am mistaken. The scrollbar dragging does work, but is inverted to the direction I expected. However, the mousewheel does not appear to work.

embeddedt, 2021-03-28:
The mousewheel is not intended to be used for scrolling. At the moment it is used to simulate a rotary encoder, which the default demo doesn't use. 

The scrolling is inverted because your mouse pointer is treated like a finger on a touchscreen. I should also note that unlike other mobile interfaces, dragging the scrollbar itself (as opposed to the page content) does not invert the scrolling. LVGL treats them both as if you are dragging the page.

jeremyherbert, 2021-03-28:
Ok, thank you!

closed 2021-03-28 by jeremyherbert