lvgl/lv_port_pc_eclipse #93: How do I generate an ELF file with debugging information？

jianjunxia, 2021-04-25:


embeddedt, 2021-04-25:
As far as I know, the project generates debug symbols by default. Are you using Eclipse or CMake?

jianjunxia, 2021-04-25:
> As far as I know, the project generates debug symbols by default. Are you using Eclipse or CMake?

I use vscode,I now have a problem, I can’t use gdb to break points

jianjunxia, 2021-04-25:
> As far as I know, the project generates debug symbols by default. Are you using Eclipse or CMake?
i use cmake in ubuntu_14.0, i use cmd: file main, no display with debug info
main: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=f34a523dd396d5cc3986f01d55353b46ac20ce53, for GNU/Linux 3.2.0, not stripped

embeddedt, 2021-04-25:
Try this in the root folder:

```
mkdir cmbuild
cd cmbuild
cmake -DCMAKE_BUILD_TYPE=Debug ..
make -j4
```

In my clone this produces a binary with debug symbols.

jianjunxia, 2021-04-25:
> Try this in the root folder:
> 
> ```
> mkdir cmbuild
> cd cmbuild
> cmake -DCMAKE_BUILD_TYPE=Debug ..
> make -j4
> ```
> 
> In my clone this produces a binary with debug symbols.

> Try this in the root folder:
> 
> ```
> mkdir cmbuild
> cd cmbuild
> cmake -DCMAKE_BUILD_TYPE=Debug ..
> make -j4
> ```
> 
> In my clone this produces a binary with debug symbols.

Great! Thank you for solving my problem   :）

embeddedt, 2021-04-25:
Glad to help!

closed 2021-04-25 by embeddedt