lvgl/lv_port_pc_eclipse #94: SDL2/SDL.h'

aspdigital, 2021-05-08:


embeddedt, 2021-05-08:
You can configure the exact include path in the [`lv_drv_conf.h`](https://github.com/lvgl/lv_sim_eclipse_sdl/blob/e164e3591c3e1e3bf3464d19e0dcdc67b2e6a791/lv_drv_conf.h#L101) file.

I don't have access to a Mac, so I am unfortunately unable to help much further (hopefully someone else can!), however, from some quick and limited research:

* `<SDL.h>` looks like the proper include name to use.
* `/Library/Frameworks/SDL2.framework/Headers` needs to be added to your Eclipse include path.

That should make compilation succeed but you might run into more errors when it gets to the linking stage.

aspdigital, 2021-05-08:
I put a symlink in /Library/Frameworks called SDL2 which links to SDL2.framework.Headers 

and that somewhat resolves the problem. Now I get:

warning: 'MONITOR_SDL_INCLUDE_PATH' macro redefined

in the various disabled display driver headers.

As expected, there are linker issues:

ld: library not found for -lSDL2main

I see that SDL2main is called out in my C/C++ Build -> Settings -> Linker -> Libraries, but I have no idea where it's supposed to live, or what builds it, etc.

I was hoping to not go down such a rabbit hole.

kisvegabor, 2021-05-10:
Have you installed SDL2 with `brew install sdl2`? 
The last time when I tested on Mac it worked with any extra modifications.

closed 2021-06-26 by stale[bot]