lvgl/lv_port_pc_eclipse #95: Simulator code with LVGL v6

kashishkhullar, 2021-06-03:
### Need release with v6 of lvgl
I am using lgvl v6 library for my project and I want to run my code on a simulator due to many reasons. But I noticed that you haven't provided any simulator boilerplate for v6 of lvgl. Is there any reason why v6 for skipped completely?
If I want to create a simulator for my project, what are options here? Given that upgrading to a higher version of lvgl is not possible for me right now.

embeddedt, 2021-06-03:
The tag seems to have disappeared/was never created. However, you can use this commit: https://github.com/lvgl/lv_sim_eclipse_sdl/commit/00ed6e1a74414abfc168eef8eac59fdb896c5e72, and manually update the lvgl submodule to the `release/v6` branch so you get the fixes since 6.1 was released.

embeddedt, 2021-06-26:
Closing as the question was answered.

closed 2021-06-26 by embeddedt