lvgl/lv_port_pc_eclipse #96: Error when using make - fatal error: lvgl/lvgl.h: No such file or directory

qassim93, 2021-06-26:
Hi,

First I would like to thank you for your great efforts and the open source lvgl library. 

I've tried using the demo on windows simulator using code blocks and everything seems to be working fine.

My target is to get the GUI running on a RPi 3B+ so I tried to run the simulator on it. I've followed many instructions and so from github and lvgl forums but I don't seem to be able to get working. I'm not sure if it is hard to get lvgl up and running on RPi or if I don't have the skills and knowledge to do so.

I tried the instructions mentioned in the below link, but I get an error that the lvgl.h dosnt exist. I tried to manually copy the content of lvgl file from another file but I get many errors afterwards. Could you please help me with this?
link to the instructions I've followed: https://github.com/lvgl/lvgl/issues/380

Error I get:
pi@raspberrypi:~/Desktop/test3/pc_simulator/build $ make
Scanning dependencies of target main
[ 33%] Building C object CMakeFiles/main.dir/main.c.o
/home/pi/Desktop/test3/pc_simulator/main.c:15:10: fatal error: lvgl/lvgl.h: No such file or directory
 #include "lvgl/lvgl.h"
          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/main.dir/build.make:63: CMakeFiles/main.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:77: CMakeFiles/main.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

embeddedt, 2021-06-26:
Have you updated submodules (`git submodule update --init --recursive` in the base directory)?

qassim93, 2021-06-26:
Thank you very much for your answer. 

I tried using "git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git" instead of using "git clone https://github.com/littlevgl/pc_simulator_sdl_eclipse.git" and that worked for me.

I'm new to lunix and didn't pay attention to "--recursive". I need to look it up and understand these commands better. 

The example works perfectly on my RPi now. Now I have to find a way to add my (.c) and (.h) files that I've created in code blocks on windows.

kisvegabor, 2021-06-28:
@jojomojo93 
Glad to hear it works now. Note that these `git` commands are the same on every OS. 

As the issue is solved I close it now.

closed 2021-06-28 by kisvegabor