lvgl/lv_port_pc_eclipse #97: No "Debug" forlder - No simulator - Linux Eclipse CDT

mouffa, 2021-08-01:
Hello, I am trying to simulate LVGL with SDL on Ubuntu 21.10 with Eclipse CDT, everything works fine but I can not find any "Debug" folder in the repository so after recursive cloning of the repository there is no "Debug" folder in the project too, thus the Eclipse can not find the "pc_simulator" binary. The code compiles successfully.

embeddedt, 2021-08-01:
Try cleaning and rebuilding in Eclipse; in my experience, sometimes it takes two tries to actually create the folder.

mouffa, 2021-08-01:
I thought it was building by hitting the hammer but it was not, with right-click and Build on the project folder it builds with 0 errors and 19 warnings concerning the demos, the Debug folder appears but I can not see any "pc_simulator" binary so I get the same missing file error when I hit the Play button, as I can understand the simulator should be built when I built the project.

embeddedt, 2021-08-01:
Right click and build a second time; that is the glitch I was mentioning. If the Debug folder does appear, and Eclipse claims things have "built successfully", but there is no `pc_simulator` binary, it usually means that you have to build it again.

As far as I know this is an issue with Eclipse's build system and I'm not quite sure how to prevent it on our end.

mouffa, 2021-08-01:
It was solved but not with the second build which returned "Nothing to build", I did right-click on the project folder and "Refresh F5", then the binary appeared with a different name, that is "pc_simulator_sdl_eclipse" instead of "pc_simulator_eclipse_sdl" so I loaded it through the settings, and now the simulator works the same as running the program natively on Pinephone Arch Linux Barebone where I do the development.

embeddedt, 2021-08-01:
Hmm, okay, let me double-check with a fresh copy of the project. You shouldn't need to adjust the name; I've never had that occur before.

embeddedt, 2021-08-01:
I could not reproduce this. When cloning a fresh copy of the project in a different directory and importing it into my Eclipse 2018-12 workspace, the run configuration appears to be correct:

[image]

mouffa, 2021-08-01:
when I did the refresh after the build the simulator binary in the Debug folder had a different name than the configuration, the configuration was expected to be, so I had to change the configuration or rename the file

closed 2022-04-28 by stale[bot]