lvgl/lv_port_pc_eclipse #101: -lpng not found

buktemirlnk, 2021-12-24:
First of all, hello everyone. I am very new to LVGL and I am getting the following build errors. I am using **Ubuntu 21.04.** 
Can you help with this issue? 
```

Building file: ../mouse_cursor_icon.c
Invoking: Cross GCC Compiler
gcc -I"/home/buket/for-lvgl-tests/lv_sim_eclipse_sdl-master" -O0 -g3 -Wall -Wextra -c -fmessage-length=0 -Wall -Wmaybe-uninitialized -fsanitize=address -fsanitize=undefined -fno-omit-frame-pointer -MMD -MP -MF"mouse_cursor_icon.d" -MT"mouse_cursor_icon.o" -o "mouse_cursor_icon.o" "../mouse_cursor_icon.c"
Finished building: ../mouse_cursor_icon.c
 

Building target: lv_sim_eclipse_sdl-master
Invoking: Cross GCC Linker
gcc -fsanitize=address -fsanitize=undefined -fno-omit-frame-pointer -o "lv_sim_eclipse_sdl-master"  ./lvgl/tests/unity/unity.o ./lvgl/tests/unity/unity_support.o  ./lvgl/tests/src/test_fonts/font_1.o ./lvgl/tests/src/test_fonts/font_2.o ./lvgl/tests/src/test_fonts/font_3.o  ./lvgl/tests/src/test_cases/_test_template.o ./lvgl/tests/src/test_cases/test_arc.o ./lvgl/tests/src/test_cases/test_checkbox.o ./lvgl/tests/src/test_cases/test_config.o ./lvgl/tests/src/test_cases/test_dropdown.o ./lvgl/tests/src/test_cases/test_font_loader.o ./lvgl/tests/src/test_cases/test_obj_tree.o ./lvgl/tests/src/test_cases/test_style.o ./lvgl/tests/src/test_cases/test_switch.o ./lvgl/tests/src/test_cases/test_txt.o  ./lvgl/tests/src/lv_test_indev.o ./lvgl/tests/src/lv_test_init.o  ./lvgl/src/widgets/lv_arc.o ./lvgl/src/widgets/lv_bar.o ./lvgl/src/widgets/lv_btn.o ./lvgl/src/widgets/lv_btnmatrix.o ./lvgl/src/widgets/lv_canvas.o ./lvgl/src/widgets/lv_checkbox.o ./lvgl/src/widgets/lv_dropdown.o ./lvgl/src/widgets/lv_img.o ./lvgl/src/widgets/lv_label.o ./lvgl/src/widgets/lv_line.o ./lvgl/src/widgets/lv_objx_templ.o ./lvgl/src/widgets/lv_roller.o ./lvgl/src/widgets/lv_slider.o ./lvgl/src/widgets/lv_switch.o ./lvgl/src/widgets/lv_table.o ./lvgl/src/widgets/lv_textarea.o  ./lvgl/src/misc/lv_anim.o ./lvgl/src/misc/lv_anim_timeline.o ./lvgl/src/misc/lv_area.o ./lvgl/src/misc/lv_async.o ./lvgl/src/misc/lv_bidi.o ./lvgl/src/misc/lv_color.o ./lvgl/src/misc/lv_fs.o ./lvgl/src/misc/lv_gc.o ./lvgl/src/misc/lv_ll.o ./lvgl/src/misc/lv_log.o ./lvgl/src/misc/lv_lru.o ./lvgl/src/misc/lv_math.o ./lvgl/src/misc/lv_mem.o ./lvgl/src/misc/lv_printf.o ./lvgl/src/misc/lv_style.o ./lvgl/src/misc/lv_style_gen.o ./lvgl/src/misc/lv_templ.o ./lvgl/src/misc/lv_timer.o ./lvgl/src/misc/lv_tlsf.o ./lvgl/src/misc/lv_txt.o ./lvgl/src/misc/lv_txt_ap.o ./lvgl/src/misc/lv_utils.o  ./lvgl/src/hal/lv_hal_disp.o ./lvgl/src/hal/lv_hal_indev.o ./lvgl/src/hal/lv_hal_tick.o  ./lvgl/src/font/lv_font.o ./lvgl/src/font/lv_font_dejavu_16_persian_hebrew.o ./lvgl/src/font/lv_font_fmt_txt.o ./lvgl/src/font/lv_font_loader.o ./lvgl/src/font/lv_font_montserrat_10.o ./lvgl/src/font/lv_font_montserrat_12.o ./lvgl/src/font/lv_font_montserrat_12_subpx.o ./lvgl/src/font/lv_font_montserrat_14.o ./lvgl/src/font/lv_font_montserrat_16.o ./lvgl/src/font/lv_font_montserrat_18.o ./lvgl/src/font/lv_font_montserrat_20.o ./lvgl/src/font/lv_font_montserrat_22.o ./lvgl/src/font/lv_font_montserrat_24.o ./lvgl/src/font/lv_font_montserrat_26.o ./lvgl/src/font/lv_font_montserrat_28.o ./lvgl/src/font/lv_font_montserrat_28_compressed.o ./lvgl/src/font/lv_font_montserrat_30.o ./lvgl/src/font/lv_font_montserrat_32.o ./lvgl/src/font/lv_font_montserrat_34.o ./lvgl/src/font/lv_font_montserrat_36.o ./lvgl/src/font/lv_font_montserrat_38.o ./lvgl/src/font/lv_font_montserrat_40.o ./lvgl/src/font/lv_font_montserrat_42.o ./lvgl/src/font/lv_font_montserrat_44.o ./lvgl/src/font/lv_font_montserrat_46.o ./lvgl/src/font/lv_font_montserrat_48.o ./lvgl/src/font/lv_font_montserrat_8.o ./lvgl/src/font/lv_font_simsun_16_cjk.o ./lvgl/src/font/lv_font_unscii_16.o ./lvgl/src/font/lv_font_unscii_8.o  ./lvgl/src/extra/widgets/win/lv_win.o  ./lvgl/src/extra/widgets/tileview/lv_tileview.o  ./lvgl/src/extra/widgets/tabview/lv_tabview.o  ./lvgl/src/extra/widgets/spinner/lv_spinner.o  ./lvgl/src/extra/widgets/spinbox/lv_spinbox.o  ./lvgl/src/extra/widgets/span/lv_span.o  ./lvgl/src/extra/widgets/msgbox/lv_msgbox.o  ./lvgl/src/extra/widgets/meter/lv_meter.o  ./lvgl/src/extra/widgets/menu/lv_menu.o  ./lvgl/src/extra/widgets/list/lv_list.o  ./lvgl/src/extra/widgets/led/lv_led.o  ./lvgl/src/extra/widgets/keyboard/lv_keyboard.o  ./lvgl/src/extra/widgets/imgbtn/lv_imgbtn.o  ./lvgl/src/extra/widgets/colorwheel/lv_colorwheel.o  ./lvgl/src/extra/widgets/chart/lv_chart.o  ./lvgl/src/extra/widgets/calendar/lv_calendar.o ./lvgl/src/extra/widgets/calendar/lv_calendar_header_arrow.o ./lvgl/src/extra/widgets/calendar/lv_calendar_header_dropdown.o  ./lvgl/src/extra/widgets/animimg/lv_animimg.o  ./lvgl/src/extra/themes/mono/lv_theme_mono.o  ./lvgl/src/extra/themes/default/lv_theme_default.o  ./lvgl/src/extra/themes/basic/lv_theme_basic.o  ./lvgl/src/extra/others/snapshot/lv_snapshot.o  ./lvgl/src/extra/others/monkey/lv_monkey.o  ./lvgl/src/extra/libs/sjpg/lv_sjpg.o ./lvgl/src/extra/libs/sjpg/tjpgd.o  ./lvgl/src/extra/libs/rlottie/lv_rlottie.o  ./lvgl/src/extra/libs/qrcode/lv_qrcode.o ./lvgl/src/extra/libs/qrcode/qrcodegen.o  ./lvgl/src/extra/libs/png/lodepng.o ./lvgl/src/extra/libs/png/lv_png.o  ./lvgl/src/extra/libs/gif/gifdec.o ./lvgl/src/extra/libs/gif/lv_gif.o  ./lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.o ./lvgl/src/extra/libs/fsdrv/lv_fs_posix.o ./lvgl/src/extra/libs/fsdrv/lv_fs_stdio.o ./lvgl/src/extra/libs/fsdrv/lv_fs_win32.o  ./lvgl/src/extra/libs/freetype/lv_freetype.o  ./lvgl/src/extra/libs/ffmpeg/lv_ffmpeg.o  ./lvgl/src/extra/libs/bmp/lv_bmp.o  ./lvgl/src/extra/layouts/grid/lv_grid.o  ./lvgl/src/extra/layouts/flex/lv_flex.o  ./lvgl/src/extra/lv_extra.o  ./lvgl/src/draw/sw/lv_draw_sw.o ./lvgl/src/draw/sw/lv_draw_sw_arc.o ./lvgl/src/draw/sw/lv_draw_sw_blend.o ./lvgl/src/draw/sw/lv_draw_sw_img.o ./lvgl/src/draw/sw/lv_draw_sw_letter.o ./lvgl/src/draw/sw/lv_draw_sw_line.o ./lvgl/src/draw/sw/lv_draw_sw_polygon.o ./lvgl/src/draw/sw/lv_draw_sw_rect.o  ./lvgl/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.o  ./lvgl/src/draw/sdl/lv_draw_sdl.o ./lvgl/src/draw/sdl/lv_draw_sdl_arc.o ./lvgl/src/draw/sdl/lv_draw_sdl_bg.o ./lvgl/src/draw/sdl/lv_draw_sdl_blend.o ./lvgl/src/draw/sdl/lv_draw_sdl_composite.o ./lvgl/src/draw/sdl/lv_draw_sdl_img.o ./lvgl/src/draw/sdl/lv_draw_sdl_label.o ./lvgl/src/draw/sdl/lv_draw_sdl_line.o ./lvgl/src/draw/sdl/lv_draw_sdl_mask.o ./lvgl/src/draw/sdl/lv_draw_sdl_rect.o ./lvgl/src/draw/sdl/lv_draw_sdl_stack_blur.o ./lvgl/src/draw/sdl/lv_draw_sdl_texture_cache.o ./lvgl/src/draw/sdl/lv_draw_sdl_utils.o  ./lvgl/src/draw/nxp_vglite/lv_gpu_nxp_vglite.o  ./lvgl/src/draw/nxp_pxp/lv_gpu_nxp_pxp.o ./lvgl/src/draw/nxp_pxp/lv_gpu_nxp_pxp_osa.o  ./lvgl/src/draw/lv_draw.o ./lvgl/src/draw/lv_draw_arc.o ./lvgl/src/draw/lv_draw_img.o ./lvgl/src/draw/lv_draw_label.o ./lvgl/src/draw/lv_draw_line.o ./lvgl/src/draw/lv_draw_mask.o ./lvgl/src/draw/lv_draw_rect.o ./lvgl/src/draw/lv_draw_triangle.o ./lvgl/src/draw/lv_img_buf.o ./lvgl/src/draw/lv_img_cache.o ./lvgl/src/draw/lv_img_decoder.o  ./lvgl/src/core/lv_disp.o ./lvgl/src/core/lv_event.o ./lvgl/src/core/lv_group.o ./lvgl/src/core/lv_indev.o ./lvgl/src/core/lv_indev_scroll.o ./lvgl/src/core/lv_obj.o ./lvgl/src/core/lv_obj_class.o ./lvgl/src/core/lv_obj_draw.o ./lvgl/src/core/lv_obj_pos.o ./lvgl/src/core/lv_obj_scroll.o ./lvgl/src/core/lv_obj_style.o ./lvgl/src/core/lv_obj_style_gen.o ./lvgl/src/core/lv_obj_tree.o ./lvgl/src/core/lv_refr.o ./lvgl/src/core/lv_theme.o  ./lvgl/rt-thread/lv_rt_thread_port.o  ./lvgl/examples/widgets/win/lv_example_win_1.o  ./lvgl/examples/widgets/tileview/lv_example_tileview_1.o  ./lvgl/examples/widgets/textarea/lv_example_textarea_1.o ./lvgl/examples/widgets/textarea/lv_example_textarea_2.o ./lvgl/examples/widgets/textarea/lv_example_textarea_3.o  ./lvgl/examples/widgets/tabview/lv_example_tabview_1.o ./lvgl/examples/widgets/tabview/lv_example_tabview_2.o  ./lvgl/examples/widgets/table/lv_example_table_1.o ./lvgl/examples/widgets/table/lv_example_table_2.o  ./lvgl/examples/widgets/switch/lv_example_switch_1.o  ./lvgl/examples/widgets/spinner/lv_example_spinner_1.o  ./lvgl/examples/widgets/spinbox/lv_example_spinbox_1.o  ./lvgl/examples/widgets/span/lv_example_span_1.o  ./lvgl/examples/widgets/slider/lv_example_slider_1.o ./lvgl/examples/widgets/slider/lv_example_slider_2.o ./lvgl/examples/widgets/slider/lv_example_slider_3.o  ./lvgl/examples/widgets/roller/lv_example_roller_1.o ./lvgl/examples/widgets/roller/lv_example_roller_2.o ./lvgl/examples/widgets/roller/lv_example_roller_3 [... cut ...] _12_compr_az.c.o ./lv_demos/src/lv_demo_benchmark/assets/lv_font_bechmark_montserrat_16_compr_az.c.o ./lv_demos/src/lv_demo_benchmark/assets/lv_font_bechmark_montserrat_28_compr_az.c.o  ./lv_demos/src/lv_demo_benchmark/lv_demo_benchmark.o  ./main.o ./mouse_cursor_icon.o   -lSDL2main -lm -lpng -lSDL2
/bin/ld: -lpng bulunamadı
collect2: error: ld returned 1 exit status
make: *** [makefile:172: lv_sim_eclipse_sdl-master] Error 1
"make all" terminated with exit code 2. Build might be incomplete.
```

embeddedt, 2021-12-24:
You probably need to run `apt install libpng-dev`.

buktemirlnk, 2021-12-24:
Thank you for your answer. Now I'm getting this error :
```
Building target: lv_sim_eclipse_sdl-master
Invoking: Cross GCC Linker
gcc -fsanitize=address -fsanitize=undefined -fno-omit-frame-pointer -o "lv_sim_eclipse_sdl-master"  ./lvgl/tests/unity/unity.o ./lvgl/tests/unity/unity_support.o  ./lvgl/tests/src/test_fonts/font_1.o ./lvgl/tests/src/test_fonts/font_2.o ./lvgl/tests/src/test_fonts/font_3.o  ./lvgl/tests/src/test_cases/_test_template.o ./lvgl/tests/src/test_cases/test_arc.o ./lvgl/tests/src/test_cases/test_checkbox.o ./lvgl/tests/src/test_cases/test_config.o ./lvgl/tests/src/test_cases/test_dropdown.o ./lvgl/tests/src/test_cases/test_font_loader.o ./lvgl/tests/src/test_cases/test_obj_tree.o ./lvgl/tests/src/test_cases/test_style.o ./lvgl/tests/src/test_cases/test_switch.o ./lvgl/tests/src/test_cases/test_txt.o  ./lvgl/tests/src/lv_test_indev.o ./lvgl/tests/src/lv_test_init.o  ./lvgl/src/widgets/lv_arc.o ./lvgl/src/widgets/lv_bar.o ./lvgl/src/widgets/lv_btn.o ./lvgl/src/widgets/lv_btnmatrix.o ./lvgl/src/widgets/lv_canvas.o ./lvgl/src/widgets/lv_checkbox.o ./lvgl/src/widgets/lv_dropdown.o ./lvgl/src/widgets/lv_img.o ./lvgl/src/widgets/lv_label.o ./lvgl/src/widgets/lv_line.o ./lvgl/src/widgets/lv_objx_templ.o ./lvgl/src/widgets/lv_roller.o ./lvgl/src/widgets/lv_slider.o ./lvgl/src/widgets/lv_switch.o ./lvgl/src/widgets/lv_table.o ./lvgl/src/widgets/lv_textarea.o  ./lvgl/src/misc/lv_anim.o ./lvgl/src/misc/lv_anim_timeline.o ./lvgl/src/misc/lv_area.o ./lvgl/src/misc/lv_async.o ./lvgl/src/misc/lv_bidi.o ./lvgl/src/misc/lv_color.o ./lvgl/src/misc/lv_fs.o ./lvgl/src/misc/lv_gc.o ./lvgl/src/misc/lv_ll.o ./lvgl/src/misc/lv_log.o ./lvgl/src/misc/lv_lru.o ./lvgl/src/misc/lv_math.o ./lvgl/src/misc/lv_mem.o ./lvgl/src/misc/lv_printf.o ./lvgl/src/misc/lv_style.o ./lvgl/src/misc/lv_style_gen.o ./lvgl/src/misc/lv_templ.o ./lvgl/src/misc/lv_timer.o ./lvgl/src/misc/lv_tlsf.o ./lvgl/src/misc/lv_txt.o ./lvgl/src/misc/lv_txt_ap.o ./lvgl/src/misc/lv_utils.o  ./lvgl/src/hal/lv_hal_disp.o ./lvgl/src/hal/lv_hal_indev.o ./lvgl/src/hal/lv_hal_tick.o  ./lvgl/src/font/lv_font.o ./lvgl/src/font/lv_font_dejavu_16_persian_hebrew.o ./lvgl/src/font/lv_font_fmt_txt.o ./lvgl/src/font/lv_font_loader.o ./lvgl/src/font/lv_font_montserrat_10.o ./lvgl/src/font/lv_font_montserrat_12.o ./lvgl/src/font/lv_font_montserrat_12_subpx.o ./lvgl/src/font/lv_font_montserrat_14.o ./lvgl/src/font/lv_font_montserrat_16.o ./lvgl/src/font/lv_font_montserrat_18.o ./lvgl/src/font/lv_font_montserrat_20.o ./lvgl/src/font/lv_font_montserrat_22.o ./lvgl/src/font/lv_font_montserrat_24.o ./lvgl/src/font/lv_font_montserrat_26.o ./lvgl/src/font/lv_font_montserrat_28.o ./lvgl/src/font/lv_font_montserrat_28_compressed.o ./lvgl/src/font/lv_font_montserrat_30.o ./lvgl/src/font/lv_font_montserrat_32.o ./lvgl/src/font/lv_font_montserrat_34.o ./lvgl/src/font/lv_font_montserrat_36.o ./lvgl/src/font/lv_font_montserrat_38.o ./lvgl/src/font/lv_font_montserrat_40.o ./lvgl/src/font/lv_font_montserrat_42.o ./lvgl/src/font/lv_font_montserrat_44.o ./lvgl/src/font/lv_font_montserrat_46.o ./lvgl/src/font/lv_font_montserrat_48.o ./lvgl/src/font/lv_font_montserrat_8.o ./lvgl/src/font/lv_font_simsun_16_cjk.o ./lvgl/src/font/lv_font_unscii_16.o ./lvgl/src/font/lv_font_unscii_8.o  ./lvgl/src/extra/widgets/win/lv_win.o  ./lvgl/src/extra/widgets/tileview/lv_tileview.o  ./lvgl/src/extra/widgets/tabview/lv_tabview.o  ./lvgl/src/extra/widgets/spinner/lv_spinner.o  ./lvgl/src/extra/widgets/spinbox/lv_spinbox.o  ./lvgl/src/extra/widgets/span/lv_span.o  ./lvgl/src/extra/widgets/msgbox/lv_msgbox.o  ./lvgl/src/extra/widgets/meter/lv_meter.o  ./lvgl/src/extra/widgets/menu/lv_menu.o  ./lvgl/src/extra/widgets/list/lv_list.o  ./lvgl/src/extra/widgets/led/lv_led.o  ./lvgl/src/extra/widgets/keyboard/lv_keyboard.o  ./lvgl/src/extra/widgets/imgbtn/lv_imgbtn.o  ./lvgl/src/extra/widgets/colorwheel/lv_colorwheel.o  ./lvgl/src/extra/widgets/chart/lv_chart.o  ./lvgl/src/extra/widgets/calendar/lv_calendar.o ./lvgl/src/extra/widgets/calendar/lv_calendar_header_arrow.o ./lvgl/src/extra/widgets/calendar/lv_calendar_header_dropdown.o  ./lvgl/src/extra/widgets/animimg/lv_animimg.o  ./lvgl/src/extra/themes/mono/lv_theme_mono.o  ./lvgl/src/extra/themes/default/lv_theme_default.o  ./lvgl/src/extra/themes/basic/lv_theme_basic.o  ./lvgl/src/extra/others/snapshot/lv_snapshot.o  ./lvgl/src/extra/others/monkey/lv_monkey.o  ./lvgl/src/extra/libs/sjpg/lv_sjpg.o ./lvgl/src/extra/libs/sjpg/tjpgd.o  ./lvgl/src/extra/libs/rlottie/lv_rlottie.o  ./lvgl/src/extra/libs/qrcode/lv_qrcode.o ./lvgl/src/extra/libs/qrcode/qrcodegen.o  ./lvgl/src/extra/libs/png/lodepng.o ./lvgl/src/extra/libs/png/lv_png.o  ./lvgl/src/extra/libs/gif/gifdec.o ./lvgl/src/extra/libs/gif/lv_gif.o  ./lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.o ./lvgl/src/extra/libs/fsdrv/lv_fs_posix.o ./lvgl/src/extra/libs/fsdrv/lv_fs_stdio.o ./lvgl/src/extra/libs/fsdrv/lv_fs_win32.o  ./lvgl/src/extra/libs/freetype/lv_freetype.o  ./lvgl/src/extra/libs/ffmpeg/lv_ffmpeg.o  ./lvgl/src/extra/libs/bmp/lv_bmp.o  ./lvgl/src/extra/layouts/grid/lv_grid.o  ./lvgl/src/extra/layouts/flex/lv_flex.o  ./lvgl/src/extra/lv_extra.o  ./lvgl/src/draw/sw/lv_draw_sw.o ./lvgl/src/draw/sw/lv_draw_sw_arc.o ./lvgl/src/draw/sw/lv_draw_sw_blend.o ./lvgl/src/draw/sw/lv_draw_sw_img.o ./lvgl/src/draw/sw/lv_draw_sw_letter.o ./lvgl/src/draw/sw/lv_draw_sw_line.o ./lvgl/src/draw/sw/lv_draw_sw_polygon.o ./lvgl/src/draw/sw/lv_draw_sw_rect.o  ./lvgl/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.o  ./lvgl/src/draw/sd [... cut ...] hmark/assets/lv_font_bechmark_montserrat_28_compr_az.c.o  ./lv_demos/src/lv_demo_benchmark/lv_demo_benchmark.o  ./main.o ./mouse_cursor_icon.o   -lSDL2main -lm -lpng -lSDL2
/bin/ld: ./main.o: in function `main':
/home/buket/for-lvgl-tests/lv_sim_eclipse_sdl-master/Debug/../main.c:99: undefined reference to `lv_demo_widgets'
collect2: error: ld returned 1 exit status
make: *** [makefile:172: lv_sim_eclipse_sdl-master] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

```

closed 2022-04-28 by stale[bot]